bengtmartensson/IrScrutinizer #1: rxtx status

leamas, 2015-03-22:
The rxtx sources are from an unknown source. The diff against the original 2.2 sources at qbang.org is quite large (40 files changed, 3 insertions(+), 8595 deletions) 

In combination with a missing history, this code is hard to review. Could you possibly add some info from where these sources are forked?

bengtmartensson, 2015-03-22:
The sources are from Fedora package rxtx-2.2-0.10.20100211.fc21.src.rpm, without the Fedora patches, and a bunch of unneeded files nuked. To my understanding, the "latest official" release was 2.2pre2, while the previous mentioned sources appear somewhat newer than 2.2pre2, for example RXTXVersion.Version says RXTX-2.2, but in the released version  RXTX-2.2pre2  (the former btw clashes with the C code). 

Please note that the need of a private version arises since the Fedora RXTXcomm.jar is (deliberately) unusable for other systems. For Fedora, the packaged RXTXcomm.jar (package rxtx) is to be used, not this one. So you don't need to review :-).

Constructive suggestions are welcome. Would anyone be happy if I add the unused files?

leamas, 2015-03-22:
Anyone looking at this code needs to  understand it. That said, the need for a generic version is obvious.

I see no point in re-adding the nuked files. However, I will submit a merge request which reconstructs the history from the upstream sources.

closed 2015-03-22 by leamas