bengtmartensson/IrScrutinizer #2: so-files in git tree

leamas, 2015-03-22:
So-files (pre-compiled sources) is normally not considered as git artifacts - a git tree is about sources. Could you please consider removing these files and instead add them when creating a release e. g. as described in https://github.com/blog/1547-release-your-software?

bengtmartensson, 2015-03-22:
Disagree. "Normally" indicates a rule with exceptions. These files are:
- Not expected to change "very often", or more likely, not at all,
- At least for some, I cannot create them (Windows compiles) (from that point of view, they are "sources")

If I "had" to remove them, I would have to store these artifacts in another version control system, defeating the purpose of Github.

That said, the README could stand some improvement :-).

leamas, 2015-03-22:
Well, you certainly don't have to remove them , it's your repo. 

That said, a possible middle ground could be to let them be there for now, but  re-evaluate the situation after the first release (in which case they are available in the release tarballs).

BTW, you should perhaps mark for which arm architecture the LInux-arm file is for (32bit v5?).

I have a README update at hand, but will send a merge request before going into that.

bengtmartensson, 2015-03-25:
The ARM file was nuked (old anyhow). Looking forward for your README.

closed 2015-03-25 by bengtmartensson

leamas, 2015-03-25:
Which heavily depends on last merge request, which to a large extent is about building the libs.

commit f2ffe6c1 (2025-04-13): Change http://www.harctoolbox.org to https://...

Do not change the names of the name spaces.
Resolves #2.