bengtmartensson/IrScrutinizer #20: Replace RXTX

bengtmartensson, 2015-04-03:
RXTX is a never ending source of problems, both in porting, installation, and reliability. It is essentially very old code, that appears never to have been finished. 

There are alternatives on the Internet. First, Arduino IDE should be inspected; they migrated away from RXTX recently; is there anything there to "steal"?

bengtmartensson, 2016-06-11:
Some alternatives: JSSC, nrjavaserial (Openremote migrated RXTX-> nrjavaserial)

HelenFoster, 2016-07-18:
Arduino migrated to JSSC, and seem to be having much less trouble with serial. JSSC is also packaged in Fedora and Debian. Looks like no-one is maintaining it though.
https://github.com/scream3r/java-simple-serial-connector
https://github.com/arduino/Arduino/commit/f334e67
https://github.com/arduino/Arduino/commit/48c1223

bengtmartensson, 2016-07-19:
Thank you for the comment.

The main difference is, I think, that JCCS does not do any locking of the device file at all. (So therefore there are less problems?). Does device locking still make sense, or is it an artifact from the VAX  age? nrjavaserial still have [issues](https://github.com/NeuronRobotics/nrjavaserial/issues/60).

HelenFoster, 2016-07-19:
Not doing any locking makes the serial access more portable and avoids a large proportion of the problems.

It appears that Arduino with JSSC bypasses serial port locks placed by other applications. Since it is very popular software, this makes such locks ineffective in general, so the Arduino team have basically decided for the world that serial port locking is dead.

HelenFoster, 2016-07-20:
Testing in Windows 10, only one application can connect to the serial port, whether the applications are using RXTX, JSSC or a mixture. Probably the locking is done by the OS, though I haven't found any page stating this explicitly.

bengtmartensson, 2017-11-09:
Replaced by https://github.com/bengtmartensson/HarcHardwareBundle/issues/1. in the vein of #215.

closed 2017-11-09 by bengtmartensson