bengtmartensson/IrScrutinizer #29: irscrutinizer + gnome: inconsistent tray icon and label

leamas, 2015-04-10:
When invoking irscrutininizer using gnome, the tray icon does not match the desktop icon causing some minimal confusion. Also, the label is basically "bad"
[image]

bengtmartensson, 2015-04-10:
There are three problems here:
1. The icon as shown is the wrong one
2. It is too large
3. the label text is silly.

To 1: "This cannot happen". I cannot reproduce it either. The "antenna" icon (was remote.png in the *old* crystal icons) is not present in the Git repository. BTW, the code that sets the icon is in GuiMain.java, line 585.  I must ask you to investigate youself.

2. Probably since I set it to just one large icon, and Gnome for some reason does not scale properly. BTW, it looks perfekt on Cinnamon.

3. Also not present on cinnamon, although I can reproduce it with Gnome. There is nothing in my code that constructs "org-harctoolbox-irscutinizer-IrScrutinizer", so it must be Gnome.

leamas, 2015-04-11:
As for 2) it's the same size as other apps, so this is at least nothing odd in irscrutinizer. Leaves us with 1) and 3).

bengtmartensson, 2015-04-11:
Comment to your comment: Then it is Gnome that is broken. Cutting top an bottom of an icon is normally not considered ok...

Remains 3 (again, I do not consider 1 as a valid bug.) There are other java programs in Fedora package (at least "arduino") that exhibit the same behavior. But I cannot find out why Gnome ignores the title and somehow fumbles together something else. The program wakeonlan that you package recently says "wakeonlan" as its icon title.  (You may like to raise the same issue for arduino and see if that finds a solution :-).)

But frankly, IrScrutinizer is made to be portable and look usable (not perfect) on all systems, so I this is not a must-fix.

leamas, 2015-04-12:
> But frankly, IrScrutinizer is made to be portable and look usable (not perfect) on all systems, so I this is not a must-f

Agreed

bengtmartensson, 2015-06-27:
Closing, since it is not clear if it is a bug with the current program, or a quirk in Gnome,   I do not care very much, for reasons previously described. 

If someone contributes a fix, please reopen.

closed 2015-06-27 by bengtmartensson