bengtmartensson/IrScrutinizer #31: Crash when trying to import from irdb

probonopd, 2015-06-27:
Trying to import from irdb -> MAGIC LIGHTING -> Remotes -> Import

leads to

```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.apple.laf.AquaMenuPainter.paintSelectedMenuItemBackground(AquaMenuPainter.java:131)
	at com.apple.laf.AquaMenuItemUI.paintBackground(AquaMenuItemUI.java:110)
	at com.apple.laf.AquaMenuPainter.paintMenuItem(AquaMenuPainter.java:160)
	at com.apple.laf.AquaMenuItemUI.paintMenuItem(AquaMenuItemUI.java:83)
	at javax.swing.plaf.basic.BasicMenuItemUI.paint(BasicMenuItemUI.java:452)
	at com.apple.laf.AquaMenuItemUI.update(AquaMenuItemUI.java:99)
	at javax.swing.JComponent.paintComponent(JComponent.java:760)
	at javax.swing.JComponent.paint(JComponent.java:1037)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5106)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4890)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:814)
	at javax.swing.RepaintManager$3.run(RepaintManager.java:802)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:802)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:745)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:725)
	at javax.swing.RepaintManager.access$1000(RepaintManager.java:46)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1680)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:715)
	at java.awt.EventQueue.access$400(EventQueue.java:82)
	at java.awt.EventQueue$2.run(EventQueue.java:676)
	at java.awt.EventQueue$2.run(EventQueue.java:674)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:685)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:296)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:211)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:196)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:188)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
```

bengtmartensson, 2015-06-27:
Fixed, Thanx. (BTW, the irdb entry is not sensible, while device number empty.)

closed 2015-06-27 by bengtmartensson

probonopd, 2015-06-27:
Thanks. I know that NULL and 0 are not cleanly distinguished in the current API. Thinking about redesigning the whole irdb API.