bengtmartensson/IrScrutinizer #32: Command keys do not work in OS X

probonopd, 2015-06-27:
On OS X 10.9 the usual command-key shortcuts for copy and paste don't work. Accidentally I found out that if I change the user interface from "Metal" to "Mac OS X", then they suddenly start working. 

Hence, it would be nice if this could be fixed or if the "Mac OS X" theme could become the default on OS X.

By the way, the Arduino IDE is also written in Java but the UI looks a bit more native than the "Mac OS X" theme (e.g., menus are at the top of the screen rather than at the top of the window).

bengtmartensson, 2015-06-27:
Seems like it is a problem with Swing and Mac. Have you searched Mac+Java-Swing lists?

probonopd, 2015-06-28:
It's working fine for me now that I have selected the "Mac OS X" skin.

bengtmartensson, 2015-06-28:
I just tried it on MacOS 10.10.3. Everything works, and for all LAFs. (Except that the desktop is kidnapping F9...)

Possibly there is a problem with 10.9. Closing, since there is nothing I can do: If you have a patch that you claim fixes a problem, please reopen.

closed 2015-06-28 by bengtmartensson