bengtmartensson/IrScrutinizer #35: Find Global Caché that was switched on after launching IrScrutinizer

probonopd, 2015-06-28:
1. Launch IrScrutinizer
2. Switch on Global Caché WF2IR
3. Observe that the WF2IR shows up in the IP Name/Address dropdown
4. Press "Transmit generated" - gives HardwareUnavailableException
5. Restart IrScrutinizer - this sends a getdevices command
6. Now it is working

If IrScrutinizer would send a getdevices command as soon as it populates the IP Name/Address dropdown with the information obtained from the Beacon, then it would probably work directly without needing to restart IrScrutinizer.

bengtmartensson, 2015-06-28:
3 1/2:  You must select the desired unit explicitly from the menu.

4 1/2: Should be no need to restart, just select the desired unit.

The "getdevices" is sent as the device is selected.

probonopd, 2015-06-28:
I cannot select the desired unit explicitly from the menu because it is my only device. And it would be so much more comfortable if I didn't need to do that manually. IrScrutinizer has all the information to do it for me ;-)

bengtmartensson, 2015-06-28:
You still have to select it, even if there is only one.

> IrScrutinizer has all the information to do it for me 
No, not true; there may be more devices, that has not shown up yet.

The error message could be better, though.

probonopd, 2015-06-28:
You have a point there. Yet when just restarting IrScrutinizer it does the right thing without me having to select anything.

bengtmartensson, 2015-06-28:
> Yet when just restarting IrScrutinizer it does the right thing without me having to select anything.

Nope. You DID select it, but in the previous session :-).

probonopd, 2015-06-28:
Right, that's why I think "IrScrutinizer has all the information" :-)

probonopd, 2015-06-28:
When there is only one entry in the "IP name/Address" dropdown (e.g., populated by a Beacon), then it is not possible to select this entry (selecting it does nothing, especially it does not send a getdevices command)

bengtmartensson, 2015-06-28:
Not for me. Must be a quirk in your LAF. Try another LAF or possibly another Java, or another system.

probonopd, 2015-06-28:
Selecting different LAF does not change this. But I can live with it.

bengtmartensson, 2015-06-28:
Please send me (possible privately) a screen shot stating your point.

bengtmartensson, 2015-07-15:
@probonopd: I give you a few days more to prove that you have a point, otherwise I will close the issue.

probonopd, 2015-07-15:
[image]

bengtmartensson, 2015-07-15:
This looks bad. The primary problem (I think) is that you get an "-" in the modules list, which indicates that the response to the getdevices was not parsed correctly. The HardwareUnavailableException is then a secondary error. 

I cannot reproduce, and what the console says seems sensible. Can you set a debugger breakpoint in GlobalCache.open and see what getdevicesResult = sendCommand("getdevices", -1) does, and what getIrModules() does?

probonopd, 2015-07-15:
Module 1 and Port 1 are correctly selected (but the text is cropped in the Mac LAF). When I select another LAF the module and port are displayed correctly but the behavior is still the same. 

[image]

Below is the communication from the debug textarea. I have no clue about Java and how to set a debugger breakpoint, sorry.

```
Sending command getdevices to GlobalCache (192.168.0.15)
Connecting socket to 192.168.0.15:4998
<device,0,0 WIFI
<device,1,3 IR
<endlistdevices
Sending command getversion,0 to GlobalCache (192.168.0.15)
<1.0
1.0
Sending command getdevices to GlobalCache (192.168.0.15)
Connecting socket to 192.168.0.15:4998
<device,0,0 WIFI
<device,1,3 IR
<endlistdevices
Sending command getversion,0 to GlobalCache (192.168.0.15)
<1.0
1.0
```

bengtmartensson, 2015-07-15:
If that is the case (the LAF eats the numbers -- I may have a look at that) then I have no idea really what the problem is. You can try to delete the properties (File -> Reset properties and restart), and describe in _greatest detail_ , every single mouse click,  what you are doing.

probonopd, 2015-07-15:
OK, as I wrote everything is fine if the device is already running when I launch IRScrutinizer, so It's really a minor annoyance and I'm fine with closing this now as it is probably not worth your time.

bengtmartensson, 2015-07-16:
Was finally able to reproduce, and fix. Please try it and confirm. Thanx for finding the bug!!

A workaround for the bug is to select another sending hardware subpane ("Audio port" never makes problems...), then selecting GlobalCache again.

probonopd, 2015-07-16:
Thank you. In order to try I have to build from git using mvn, correct? Or do you plan to offer a dev build?

bengtmartensson, 2015-07-16:
Yes, you have to build it with maven, see http://www.harctoolbox.org/IrScrutinizer.html#Appendix.+Building+from+sources

probonopd, 2015-07-19:
I tried to compile this in order to test. I ahve to admit that I am entirely new to Java. `mvn install` stops with `could not be resolved: com.hifiremote:ExchangeIR:jar:0.0.8.2`on the Mac. Unfortunately the [docs](http://www.harctoolbox.org/IrScrutinizer.html#Appendix.+Building+from+sources) just say one needs to install the dependencies, but I have no clue on how to do so on the Mac.

bengtmartensson, 2015-07-19:
There are some scripts in harctoolboxbundle/tools, e.g. build-exchangeir.sh, that automates this.

However, sourceforge (where the sources reside) seems to be pining for the fjords right now...

bengtmartensson, 2015-07-23:
The Sourceforge sources (DecodeIR and ExchangeIR) temporarily (more or less :-)) available on www.harctoolbox.org, see README.md.

probonopd, 2015-07-25:
I can confirm that IrScrutinizer from svn starts to talk to the GC as soon as the GC is switched on, but the GC does not get selected in the dropdown automatically and sending a command results in the HardwareUnavailableException.

bengtmartensson, 2015-07-25:
This is the desired behavior, as motivated previously. Not that the HardwareUnavailableException now says "No GlobalCache unit selected".

Closing.

closed 2015-07-25 by bengtmartensson