bengtmartensson/IrScrutinizer #40: Nightly/development builds, e.g., using travis-ci

probonopd, 2015-07-22:
Let's provide nightly/development builds, e.g., using travis-ci. I would have written a 'travis.yml' file just now but unfortunately SourceForge is currently down, effectively preventing me from getting the dependencies.

bengtmartensson, 2015-07-23:
Sure. You help is much appreciated.

probonopd, 2015-07-23:
Tried it, but unfortunately currently my script stops in `./tools/install-deps.sh` with `build-classpath: not found
```
cat /etc/*release
# Ubuntu 14.04.1 LTS
sudo apt-get update -qq
sudo apt-get install libtool subversion maven g++ openjdk-7-jdk -qq
git clone https://github.com/bengtmartensson/harctoolboxbundle.git
cd harctoolboxbundle
./tools/build-decodeir.sh 
./tools/build-exchangeir.sh 
./tools/build-jcommander.sh
./tools/build-minimal-json.sh 
sudo mkdir -p /etc/alternatives/java_sdk/jre/lib/amd64/
/etc/alternatives/java_sdk/jre/lib/*/libjava.so 
/usr/lib/amd64/libjava.so
sudo ln -s /usr/lib/amd64/libjava.so /etc/alternatives/java_sdk/jre/lib/amd64/
./tools/build-tonto.sh 
./tools/build-wakeonlan.sh 
sudo ./tools/install-deps.sh 
# ./tools/install-deps.sh: 13: ./tools/install-deps.sh: build-classpath: not found
# [ERROR] Failed to execute goal org.apache.maven.plugins:maven-install-plugin:2.3:install-file (default-cli) on project HarctoolboxBundle: Error installing artifact 'com.hifiremote:DecodeIRCaller:jar': Failed to install artifact com.hifiremote:DecodeIRCaller:jar:2.44: /home/me/harctoolboxbundle (Is a directory) -> [Help 1]
```

bengtmartensson, 2015-07-24:
install-deps.sh was written by Alec Leamas. I am still not really sure what it does. Somewhat on the nasty side, I think basically it installs the dependencies provided that they are installed already. When it did not do what I wanted, I wrote the build-*.sh scripts. 

Bottom line: do not use install-deps.sh if you are using build-*.sh.

probonopd, 2015-07-24:
See https://github.com/probonopd/harctoolboxbundle/blob/master/.travis.yml and https://travis-ci.org/probonopd/harctoolboxbundle

probonopd, 2015-07-25:
See https://github.com/probonopd/harctoolboxbundle/blob/master/.travis.yml and https://travis-ci.org/probonopd/harctoolboxbundle

It's working now. Every commit triggers a build which will be uploaded to GitHub Releases.

probonopd, 2015-07-26:
...was working, until I pulled the newest commits from your repo, now the build on travis-ci fails with
```
[INFO] --- maven-compiler-plugin:2.3.2:compile (default-compile) @ IrScrutinizer ---
[INFO] Compiling 106 source files to /home/travis/build/probonopd/harctoolboxbundle/harctoolboxbundle/IrScrutinizer/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] bootstrap class path not set in conjunction with -source 1.5
/home/travis/build/probonopd/harctoolboxbundle/harctoolboxbundle/IrScrutinizer/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java:[1960,116] error: cannot find symbol
[ERROR]  variable properties of type Props
/home/travis/build/probonopd/harctoolboxbundle/harctoolboxbundle/IrScrutinizer/src/main/java/org/harctoolbox/irscrutinizer/GuiMain.java:[1990,111] error: cannot find symbol
[INFO] 2 errors 
```
Do you have any clue what might cause this?

bengtmartensson, 2015-07-27:
Sorry for that. Development work for issue #24 was incompletely checked in. Fixed with last commit. But, that is exactly what CI is for :-). 

Thanx.

Please make separate, new, issues when I goof up again ;-).

probonopd, 2015-07-27:
Thanks. Of course I was suspecting the error to be on my side ;-) that being said, continuous builds seem to be coming along nicely.

bengtmartensson, 2015-08-17:
Please try 7a0c197 (latest). Might need to apt-get readline-devel or such.

probonopd, 2015-08-17:
Works, thanks! travis-ci for the first time reported no errors anymore.

bengtmartensson, 2015-08-18:
Kewl. Two questions:

1. what problem do you address with ln -s /.../libjava.so stuff? And the || echo "..."?  What are the error message you get if you do not do that?

2. Do you see any possibility to generate the Windows setup.exe file? This amounts to fire up the program Inno Installer (which is OS software) with the file IrScrutinizer/target/IrScrutinizer_inno.iss as argument. On a Windows host, or possibly with some W-emulation.

You may like to use the current versions of build-decodeir.sh and build-exchangeir.sh.

probonopd, 2015-08-18:
Without the ` || echo "continuing"`I get a compilation error if all parts of tonto are not correctly built, which was the case at the time when I wrote the `.travis.yml`. However it built enough of tonto so that this was not an issue. Possibly we could fix the tonto build script instead.

I need to `sudo ln -s /usr/lib/amd64/libjava.so /etc/alternatives/java_sdk/jre/lib/amd64/` because otherwise it is not found. Ubuntu apparently looks in `/etc/alternatives/java_sdk`. Possibly we could use an environment variable (which?) to set another path instead.

bengtmartensson, 2015-08-18:
Try to call build-tonto.sh with the -n flag. That does not build the shared lib. Should eliminate the need of both fixes.

Building the setup.exe is deeply integrated in the building process, so it cannot just be made on another service. What would be needed is something like

    windowsemu C:\Program Files\Inno Setup\inno.exe"  IrScrutinizer/target/IrScrutinizer_inno.iss

(whatever windowsemu means) within the same build. Possible?

Your build-decodeir.sh and build-exchageir.sh still contain the sourceforge outage fix. Mine do not. But not a big deal.

probonopd, 2015-08-18:
`-n` allowed me to remove the workaround for tonto indeed.

As for Windows, are you saying the jar cannot be built on Windows itself but only the installer is made on Windows, using the build artefacts from Linux?

If it can be done on Windows entirely, where do I find instructions on how to build the dependencies on Windows? (I suppose there must be `.bat` files involved instead of `.sh` files?)

bengtmartensson, 2015-08-18:
I do not think it is a good idea to build on Windows. Main reason is that it is a bad build environment. The backslashes and the drive letters are just the beginning....

It is just one program (c:\Program Files\Inno Setup 5\ISCC.exe to be exact) that is invoked. It essentially bundles the built artifacts into a "setup.exe" (IrScrutinizer-_VERSION_.exe). The coolest thing would be to create a portable (Java, Python,...) version of ISCC.exe, but that is probably not done during a coffee break. 

I just checked in a new implementation of the MacOS app builder (see #51). Please try it. So you do not need to generate it separately, at least not with the same file name.

bengtmartensson, 2015-08-21:
FWIW: [Here](http://news.jrsoftware.org/read/article.php?id=102449&group=jrsoftware.innosetup#102449) I have posted a question on Inno on -non-windows.

bengtmartensson, 2015-09-09:
In the stuff just committed as release 1.1.3, Wine runs ISCC just fine (see IrScrutinizer/pom.xml). Any chance you can travis-ify this?

probonopd, 2015-09-12:
Duh. Wine on travis. Well. Is the Inno stuff really needed in the first place?

probonopd, 2015-11-18:
Is there anything I can help with so that this project can start having at least Linux continuous builds?

bengtmartensson, 2015-11-19:
Thank you for the offer. I have made you a "collaborator", so you should be able to set it up.

Or course, building the "generic binary" is the top priority, everything else is secondary. 

WIne and Inno is used only for the Windows setup.exe, and can be left out. Note that the IrScrutinizer pom.xml cleanly omits it if ${basedir}/../../Inno Setup 5/ISCC.exe is missing (line 61).

Please add the used python scripts to the tools folder, to make things self-contained.

Again, thank you!

probonopd, 2015-11-27:
I added `.travis.yml` file. Next, you need to sign up at https://travis-ci.org/ and set it up to build `bengtmartensson/harctoolboxbundle`.

bengtmartensson, 2015-12-03:
@probonopd :
Thanx. Did that. And I installed your travis2githib.py. It _almost_ works in the sense that when an existing release is deleted, the tag remains, so that the newly created release will still point to the same commit, which is a bit confusing since the tag and the source zip and tarball does not agree with the uploaded binary products. Tried to delete that tag (line 59 in my version of travis2github), but it did not seem to have the desired effect, and seem to trigger additional builds.

probonopd, 2015-12-03:
You are correct, that is not ideal yet. Also travis2github is spamming the GitHub notifications feed of other users a bit. Possibly GitHub Releases are not the best way to host the builds; let's think about using something like [BinTray](https://bintray.com/) which appears to be free for Open Source projects.

bengtmartensson, 2015-12-04:
I changed the message of the generated release to try to clarify. 

Since the task has been achieved, although with a beauty spot, I consider it done and will close the issue. If you have further suggestions or improvements, please reopen or open a new issue.

Again, thanx for your help!!

closed 2015-12-05 by bengtmartensson