bengtmartensson/IrScrutinizer #44: Consider using packr for bundling the app

probonopd, 2015-07-26:
Consider https://github.com/libgdx/packr for generating the Mac(and possibly other OSes) native app.

bengtmartensson, 2015-07-26:
Thanx for the suggestion.

At least for the time being, I do not consider "native applications" (packages with included JVM) to be a good idea, Instead, I try to provide a _truly portable_ program.

But feel free to do so yourself.

closed 2015-07-26 by bengtmartensson

probonopd, 2015-07-26:
Understand what you are saying. I like your .app though. Feels so much more native on a Mac.

bengtmartensson, 2015-07-27:
Nice to hear that. BTW, building the Mac app is a PITA, see  http://www.harctoolbox.org/IrScrutinizer.html#Mac+OS+X+app+creation, but since the main problem is the Apple copyright, I do not see a simple solution.

probonopd, 2015-07-27:
I think there might be ways around that by writing your own app stub or finding a suitable existing one.

bengtmartensson, 2015-07-27:
This cannot be a new issue, any open source developer for Apple systems must have run into it. Possibly there are already solutions out there? (Apple development forums...)

Personally, it is not on the top of my priorities...

probonopd, 2015-07-27:
Like [this](https://github.com/tofi86/universalJavaApplicationStub)?

probonopd, 2015-07-27:
Or even simpler, [this](https://github.com/probonopd/harctoolboxbundle/blob/master/package/IrScrutinizer.app/Contents/MacOS/Launcher)?

bengtmartensson, 2015-08-16:
Continued as #51.