bengtmartensson/IrScrutinizer #49: Build error: Could not find artifact org.gnu:Readline:jar

probonopd, 2015-08-14:
Unfortunately I am getting build errors again. [Full log](https://s3.amazonaws.com/archive.travis-ci.org/jobs/75603011/log.txt)

```
[WARNING] The POM for org.gnu:Readline:jar:0.8.2 is missing, no dependency information available
Downloading: http://repo.maven.apache.org/maven2/com/eclipsesource/minimal-json/minimal-json/0.9.2/minimal-json-0.9.2.jar
Downloading: http://repo.maven.apache.org/maven2/org/gnu/Readline/0.8.2/Readline-0.8.2.jar
3/24 KB   
7/24 KB   
11/24 KB   
15/24 KB   
19/24 KB   
23/24 KB   
24/24 KB   
           
Downloaded: http://repo.maven.apache.org/maven2/com/eclipsesource/minimal-json/minimal-json/0.9.2/minimal-json-0.9.2.jar (24 KB at 1011.1 KB/sec)
           
Downloading: https://oss.sonatype.org/content/repositories/releases/org/gnu/Readline/0.8.2/Readline-0.8.2.jar
           
Downloading: https://repository.apache.org/releases/org/gnu/Readline/0.8.2/Readline-0.8.2.jar
           
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] HarctoolboxBundle .................................. SUCCESS [  8.112 s]
[INFO] IrpMaster .......................................... SUCCESS [ 14.784 s]
[INFO] RXTX ............................................... SUCCESS [  3.184 s]
[INFO] HarcHardware ....................................... FAILURE [  0.884 s]
[INFO] Girr ............................................... SKIPPED
[INFO] Jirc ............................................... SKIPPED
[INFO] Crystal-Clear-Icons ................................ SKIPPED
[INFO] IrScrutinizer ...................................... SKIPPED
[INFO] harc-package ....................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 27.635 s
[INFO] Finished at: 2015-08-14T13:19:50+00:00
[INFO] Final Memory: 30M/208M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project HarcHardware: Could not resolve dependencies for project org.harctoolbox:HarcHardware:jar:0.9.3: Could not find artifact org.gnu:Readline:jar:0.8.2 in central (http://repo.maven.apache.org/maven2) -> [Help 1]
```

bengtmartensson, 2015-08-14:
Sorry for that, and thank you for the report. Try the recent commit, which gets the new dependency. You must probably add this to your travis config. The compilation of the C files are not that portable; does it work for you?

probonopd, 2015-08-14:
Thanks. It's working again :-)

closed 2015-08-14 by probonopd

bengtmartensson, 2015-08-15:
Well, I am the one you should be thanking you! But it appears as if the compilation of the shared library fails due to not found include  jni.h. So, either a ad-hoc fix (basically hard coding where jni.h (etc) is located in the travis environment), or, the issue mentioned above. Needless to say, I prefer the latter.

probonopd, 2015-08-15:
The remaining error in the build log seems to be 

```
[ERROR] Failed to execute goal on project harc-package: Could not resolve dependencies for project org.harctoolbox:harc-package:jar:1.0.0: Could not find artifact org.harctoolbox:IrScrutinizer:jar:1.1.2 in central (http://repo.maven.apache.org/maven2) -> [Help 1]
```

However, since the artifacts get built anyway, I haven't worried too much about this so far.