bengtmartensson/IrScrutinizer #57: Bugs on OS X?

z3t0, 2015-08-21:
I use OS X 10.10.4 and there seem to be a lot of bugs when right clicking on a selection and then choosing an option, for example when choosing one of the received codes under the Parametric remote option, none of the move down/up options work.

What operating system is this tested on?

probonopd, 2015-08-21:
Try changing the Look and Feel to "Mac OS X" in the menu. Some things appear to work better for me with this being set.

z3t0, 2015-08-21:
@probonopd I already have it set to this. I will try and figure out what exactly does not work and post some screenshots, along with the output (java errors).

bengtmartensson, 2015-08-22:
On my Mac (MacOS 10.10.5), it works. I cannot reproduce it. (You have not enabled the sorter?) Verified with the LAF Metal, Nimbus, Mac OS.

Generally speaking, this is a _portable_ program that runs on all Java platforms (almost). I do not really fix buggy Java-implementations. 

You testing efforts would certainly be welcome. Please supply detailed bug reports.

z3t0, 2015-08-22:
Unfortunately the output is not very useful. E.g.

When trying to move a list item down in Parametric Remote.

Unfortunately I cannot copy the errors no matter how I try. It is very buggy...

I will try and build the project locally and see if I can figure out what causes the errors, I will close the issue for now since it will be a while before I can try to work on this.

closed 2015-08-22 by z3t0

bengtmartensson, 2016-04-28:
Related #119?