bengtmartensson/IrScrutinizer #77: NPE in IrScrutinizer when exporting an exotic Pronto code (which is fine though)

mxbr, 2015-11-06:
Hi, first congratulations for this awesome program =). Using Ircrutinizer 1.1.3 when i try to export this IR Code to LIRC (Im fine with it being just raw)

0000 007E 005B 0000 0064 0064 0021 0020 0021 0020 0021 0040 0021 0020 0021 0020 0042 0040 0021 0020 0042 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 0090 0064 0021 0020 0021 0020 0021 0040 0021 0020 0021 0020 0042 0040 0021 0020 0042 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 0090 0064 0021 0020 0021 0020 0021 0040 0021 0020 0021 0020 0042 0040 0021 0020 0042 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 0042 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0020 0021 0040 0021 0020 00A5 0090

Thanks! 
```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at org.harctoolbox.girr.Remote.hasThisProtocol(Remote.java:228)
	at org.harctoolbox.irscrutinizer.exporter.LircExporter.export(LircExporter.java:76)
	at org.harctoolbox.irscrutinizer.exporter.RemoteSetExporter.export(RemoteSetExporter.java:66)
	at org.harctoolbox.irscrutinizer.exporter.RemoteSetExporter.export(RemoteSetExporter.java:84)
	at org.harctoolbox.irscrutinizer.exporter.RemoteSetExporter.export(RemoteSetExporter.java:132)
	at org.harctoolbox.irscrutinizer.exporter.RemoteSetExporter.export(RemoteSetExporter.java:125)
	at org.harctoolbox.irscrutinizer.GuiMain.saveSignal(GuiMain.java:1066)
	at org.harctoolbox.irscrutinizer.GuiMain.saveSignal(GuiMain.java:1122)
	at org.harctoolbox.irscrutinizer.GuiMain.saveSignal(GuiMain.java:1112)
```

bengtmartensson, 2015-11-06:
Thanx for the bug report. Will fix.

bengtmartensson, 2015-11-08:
Fixed by recent commit. Closing. Thanx again for helping improve IrScrutinizer!

closed 2015-11-08 by bengtmartensson