bengtmartensson/IrScrutinizer #80: Improve lirc output by analizing file before writing to disk

mxbr, 2015-11-09:
Hi, this is not really an issue but a suggestion. Would it be possible to run the lirc analyze command before writing the file to the hard disk?  Or did i miss an option to do this? 

Example:
A codeset file with codes like this:
protocol = Aiwa, device = 8, subdevice = 0, obc = 26

will result in exported lirc raw file 
```
name On/Off
				8800 4400 550 550 550 550 550 550 
				550 1650 550 550 550 550 550 550..... 
```
However if i run that file troguh 

irrecord -a lircfile.conf

I get a non-raw lirc file. I prefer non-raw because file size is much smaller and its easier to track down errors or duplicate codes.

Thanks! :)

```
begin remote

  name  unknown_thing
  bits           42
  flags SPACE_ENC|CONST_LENGTH
  eps            30
  aeps          100

  header       8800  4400
  one           550  1650
  zero          550   550
  ptrail        550
  gap          173800
  toggle_bit_mask 0x0
  frequency    38000

      begin codes
          On/Off                   0x0401DFF30CF
          Mute                     0x0401DFF58A7
          TV/AV                    0x0401DFFD827
```

bengtmartensson, 2015-11-09:
Thank you for the suggestion.

For reasons of portability (see doc/CONTRIBUTE.md) and ease of installation, I do not consider invoking the C program `irrecord` to be  option -- the cost is higher than the benefit. But there are two things that can be done:

1. The Lirc exporter (.../exporter/LircExporter) already knows two protocols (NEC1 and RC5) for which it exports non-raw codes. Extending with more protocols is doable and not that hard really. 

2. IrScrutinizer (in the form of the subproject Jirc, http://www.harctoolbox.org/Jirc.html) already contains a substantial amount of Lirc 0.9.0 transformed to Java. If someone can contribute the Java translation of the pertinent parts of irrecord, as extension to Jirc, I would be inclined to use it. A few days of work.

leamas, 2015-11-09:
One thing to consider is that lirc did a major refactoring of the irrecord code during the 0.9.3 cycle, basically transforming the monolithic irrecord to a GUI client + some library functions. The core is the same, though. So, it partly becomes a question of following upstream lirc or build on current jirc. No idea what path to take here.

IMHO there is a big value in the independent implementation of the NEC and RC5.  From a LIRC perspective I think extending it with more protocols adds more to the overall community. Comparing output from different tools gives more than basing things on the same codebase - a codebase which in lirc's case is hard to maintain and I doubt anyone really, really understand.

mxbr, 2015-11-09:
I also think an independent java implementation from scratch is more helpful. I might even help too with it, i will try to check out the project and if its not hard to build and debug, i will try. Since i wrote some lirc converters for my android project, maybe i can adapt them without rewriting too much

bengtmartensson, 2015-11-11:
Forgot to write in the first message: I do not consider the  lirc.config  to be a very "future proof" format, for a number of reasons. I hope that "Lirc 2.0" (should there ever be such a thing), will be using Girr as its storage format, or at least supporting Girr-import. For this reason, the subject of this issue is not on the top of my priority list. 

Partially for this reason, creating a Java version of irrecord -a does not appear to be a very fruitful idea -- much work and low value.

If someone contributes protocol implementations in the sense of NEC1 and RC5 above, I will be happy to integrate them. If so, I can provide a cleaned-up interface.

bengtmartensson, 2015-12-01:
Closing this for now; no activities for three weeks. It is not on my list of priorities, for reasons previously listed. Still, contributed implementations of other protocols are welcome, and will likely be integrated. Please re-open if appropriate.

closed 2015-12-01 by bengtmartensson

bengtmartensson, 2016-06-08:
My current plan is replace the Lirc exporter by a pure XSLT-implementation.

bengtmartensson, 2016-06-28:
The commit 885854e22cea9ed9d7604fe2ed521283f8dafbe1 contains the above mentioned change. Still, only NEC1 and RC5 are supported, but adding more protocols would be just to add some lines of XSLT code. (Anyone?)