bengtmartensson/IrScrutinizer #81: Port to Android

bengtmartensson, 2015-11-09:


HelenFoster, 2016-08-03:
Just wondering, which of the projects are you hoping to port to Android? E.g. an actual Android version of IrScrutinizer, or more of a deployment solution based on Girr and IrpMaster?

bengtmartensson, 2016-08-03:
I _think_ that it should be possible to do a fairly complete port of IrScrutinizer, but of course I may be wrong. The hardest this is the USB support of course. But also an incomplete port would be interesting.

HelenFoster, 2016-08-03:
I'm pretty much a beginner in Android, but I think any GUI code would need to be completely rewritten.

USB might work. I was trying to use this a few months back, but never actually managed to add the library to a project in Android Studio.
https://github.com/mik3y/usb-serial-for-android

And if the phone has built-in IR which supports raw signals, it makes sense to use that.

closed 2023-12-25 by bengtmartensson