bengtmartensson/IrScrutinizer #82: Development Instructions on how to setup Build Environment for debugging

irplus-remote, 2015-11-11:
Hi, i don't know if i missed it somewhere, but are there instructions how i can setup a build environment for irscrutinizer? I am not used to all this *.pom stuff, so a step-by-step instruction would be useful (e.g. Install eclipse, Install <plugin>, etc.)

I tried to checkout the project to fiddle with it but i wasnt able to build it ;/

bengtmartensson, 2015-11-11:
There is the [Appendix in the manual](http://www.harctoolbox.org/IrScrutinizer.html#Appendix.+Building+from+sources). Of course, not perfect. Suggestions for improving the documentation are of course welcome.

Personally, I use not Eclipse but Netbeans, but it is not required (except for GUI design).

bengtmartensson, 2015-11-17:
Anything more to add? Otherwise I close. 5, 4, 3,...

closed 2015-11-19 by bengtmartensson