bengtmartensson/IrScrutinizer #86: Confirmation exiting dialog if tables non-empty?

bengtmartensson, 2015-12-13:
Should there be a confirmation dialog before ending the program if, e,g, the tables are non-empty? Or should they quietly be backed up?

Just lost 15 minutes of work... grrr.

bengtmartensson, 2016-06-20:
There should be an option for this, saved in the properties. Ideally, a "Do not ask again" checkbox.

HelenFoster, 2016-07-18:
Long-term it would be nice to have "project files", which save the tables and anything else that makes sense. Then, exiting could be the usual "unsaved changes" procedure. I'm thinking of Audacity as an example (work on the project file and export to a standard format when finished).

bengtmartensson, 2016-07-19:
@HelenFoster : Thanx for the suggestion. Right now, many of the settings, e.g. the parameters on "Generate", are saved between the sessions in the property file. (The number of those parameters may be extended.) You can have several by using the --properties <filename> option, although that is definitely an "expert option".

Also, since IrScrutinizer is not a program that "edits a certain type of file", there is no "save" and "load", but instead a number of import and export functions. 

For this reason, I do not think "project files" fits in, at least not in the current version. Open a separate ticket if appropriate.

bengtmartensson, 2016-08-10:
Confirmation dialogs with "do not ask again" option implemented.

closed 2016-08-10 by bengtmartensson