bengtmartensson/IrScrutinizer #90: Strange behaviour when exporting RC5 to LIRC

mxbr, 2015-12-29:
Hi, i noticed something strange and i think it might be a bug:

Assume the following IR-code as pronto format:

0000 0073 0000 000B 0040 0020 0020 0020 0020 0040 0040 0020 0020 0020 0020 0020 0020 0040 0040 0020 0020 0040 0020 0020 0020 0CA8

(protocol = RC5, device = 8, obc = 83, misc = T=0)

Now I exported to LIRC 

```
begin remote
	name	IrScrutinizerExport
	bits	13
	flags	RC5|CONST_LENGTH
	eps	30
	aeps	100
	zero	889	889
	one	889	889
	plead	889
	gap	90886
	toggle_bit	2
		begin codes
			IrScrutinizer captured signal	0x00000000000253
		end codes
end remote
```

If i take this LIRC-Code and re-import it into Irscrutinizer i get the following...
Note (Importing with spaces inside the button name fails with line 19:17 required (...)+ loop did not match anything at input 'captured'
Error [<clipboard>]: Parse error occured, continuing), so i removed the spaces 

But then the IR-Code imported will be:

0000 006D 0000 000A 0044 0022 0022 0022 0022 0044 0044 0022 0022 0044 0044 0044 0044 0022 0022 0044 0022 0022 0022 09EE

->> protocol = RC5, device = 9, obc = 83, misc = T=0 

So somewhere the device code was not correctly parsed. Thanks for looking into it :)

bengtmartensson, 2015-12-29:
Thank you for the report. Two bugs at the same time...

Fixed in commit above, you can use the CI build [here](https://github.com/bengtmartensson/harctoolboxbundle/releases).

bengtmartensson, 2016-01-01:
Is the problem solved for you?

mxbr, 2016-01-01:
Hi i can confirm its working now :+1:

closed 2016-01-01 by mxbr