bengtmartensson/IrScrutinizer #92: Exception in thread "AWT-EventQueue-0" java.lang.ArithmeticException: / by zero

BartmanEH, 2016-01-05:
After considerable fiddling, I have successfully loaded GirsLite 2015-09-12 into my Arduino Uno and installed IrScrutinizer v1.1.3 in Windows 10 Pro with Java 8 Update 66 but when I run a Test, I get the following immediately after any remote IR signal:

Exception in thread "AWT-EventQueue-0" java.lang.ArithmeticException: / by zero
	at com.hifiremote.exchangeir.Analyzer.identify(Analyzer.java:1130)
	at com.hifiremote.exchangeir.Analyzer.setup(Analyzer.java:414)
	at com.hifiremote.exchangeir.Analyzer.<init>(Analyzer.java:214)
	at org.harctoolbox.IrpMaster.ExchangeIR.interpretIrSequence(ExchangeIR.java:248)
	at org.harctoolbox.IrpMaster.ExchangeIR.interpretIrSequence(ExchangeIR.java:281)
	at org.harctoolbox.IrpMaster.ExchangeIR.interpretIrSequence(ExchangeIR.java:287)
	at org.harctoolbox.irscrutinizer.GuiMain.captureTestButtonActionPerformed(GuiMain.java:7715)
	at org.harctoolbox.irscrutinizer.GuiMain.access$20000(GuiMain.java:152)
	at org.harctoolbox.irscrutinizer.GuiMain$204.actionPerformed(GuiMain.java:5102)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

bengtmartensson, 2016-01-06:
Yes, that is a demodulating receiver ("38kHz"), and will not work with current IrScrutinizer. (Cf. [http://www.hifi-remote.com/forums/viewtopic.php?t=14986&start=81](this thread) showing exactly the same symptoms,)

You may like to check out my Arduino hardware guide is coming later this week.

> After considerable fiddling...

Constructive suggestions for improvements are always welcome.

BartmanEH, 2016-01-06:
Thank you, Bengt. Most of my issues getting to this point were related to my java environment under Windows 10 and trying to use different versions of GirsLite that didn't compile with Arduino 1.6.6. Fortunately GirsLite 2015-09-12 compiled without errors.

Today I bought a LTR-3208E IR detector. I used this app note https://www.fairchildsemi.com/application-notes/AN/AN-3005.pdf as a guideline to put it in switch mode with a 5k pull-up from the collector to 5VDC and grounded the emitter. I also connected the collector to Arudino pin 8.

I'm getting better results but still get the occasional java / by zero exception, program lock up requiring task manager to kill the java platform SE and restarting IrScrutinizer, and inconsistent results with a known good remote that works perfectly with Arduino IRremote library that interprets it to have NEC protocol.  All this means that I'm unable to move on to my task of decoding an unknown remote.

Is there an improvement I can make to my IR detector circuit to help get more consistent results?

bengtmartensson, 2016-01-07:
Please use the current AGirs, version 2015-12-01. I am not willing to support old versions. If "it does not compile" (it does for me), raise an issue at AGirs.

I  recommend a TSMP58000 or similar as IR sensor. Excellent reliability, excellent range. 

If you can help me making IrScrutinizer more robust for "silly" signals, please do. Be sure to turn on Options -> verbose. For example, a statement like "after receiving the following signal: xxxxx, IrScrutinizer hangs" can be helpful.

My hardware guide is almost finished...

BartmanEH, 2016-01-08:
I downloaded AGirs 2015-12-01, copied the GirsLib folder into my Arduino IDE Libraries folder and tries to compile GirsLite.ino but get the following:
C:\Users\Bart_S\Documents\Arduino\AGirs-master\src\GirsLite\GirsLite.ino:7:27: fatal error: InfraredTypes.h: No such file or directory
#include <InfraredTypes.h>
This is why I tried the earlier version which compiled without issue. The Arduino IDE is annoying to me and dependencies in general terribly difficult to figure out.  What am I missing?

bengtmartensson, 2016-01-08:
You need to copy  [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) to your Arduino library folder.  (This is a part of the old AGirs, that has been made independent).

My hardware guide is finished, see http://www.harctoolbox.org/arduino_nano.html and http://www.harctoolbox.org/arduino_nano_part2.html , possibly in particular http://www.harctoolbox.org/arduino_nano_part2.html#Compiling+the+sources

> The Arduino IDE is annoying to me 

That makes it two of us!! I use Netbeans for editing, compilng and debugging (on PC), and the silly Arduino IDE only for compiling and uploading. 

The Arduino community's belief that all on established SW engineering should be thrown away because it is supposed to be simpler for a hypothetical beginner that way just  is not good for my health...

BartmanEH, 2016-01-10:
I've downloaded a copied Infrared4Arduino to my Arduino library but I still get and error about LiquidCrystal_I2C.h so I downloaded it and copied it into the library folder as well but I still get the final error reported above:

GirsLite.cpp:1: error: expected unqualified-id before '.' token
 ../Girs/Girs.cpp
 ^
exit status 1
expected unqualified-id before '.' token

On another note, the only IR detector available at my electronics shop was a LTR-3208E IR detector but reading its datasheet http://optoelectronics.liteon.com/upload/download/DS-50-92-0068/S_110_R3208E.pdf, it has an Emitter-Collector Voltage of 5V which makes it unsuitable in my 5V circuit, right? I think I need to wire it up differently, maybe in inverting mode, to work?

bengtmartensson, 2016-01-10:
[1] LiquidCrystal:I2C.h is not needed for Girs_Lite_. I have removed it from the *.ino file in a recent commit. Thanx.

[2] There is no way that I can understand that sort of error messages. Something is probably very broken.

[3] The "second part" mentions some alternatives. mouser.com ships them worldwide.

BartmanEH, 2016-01-10:
[2] the only contents of GirsLite.cpp is "../Girs/Girs.cpp" and the Arduino IDE cannot make any sense of this. What other information can I supply to debug this issue?

bengtmartensson, 2016-01-11:
OMG, now I understand! (Something _was_ very broken...) That file is not a one-line text file but a symbolic link, and "your" git cannot make sense of it; instead it generates a silly text file, which bites you.

The second best solution is to remove said file, and replace it with the one the symlink should point to, namely ../Girs/Girs.cpp (or, if you prefer,  ..\Girs\Girs.cpp.)

Very best solution is,... you guess it, ...

Thanx for bringing this problem to my knowledge.

BartmanEH, 2016-01-11:
OK, still several problems left for me, a lowly Windows user :-)

[1] it's not enough to comment out or remove the LiquidCrystal:I2C.h from the *.ino file: there are compile errors introduced in \Girs\Girs.cpp which I copied into the GirsLite folder:

In file included from sketch\Girs.cpp:19:0:
C:\Users\Bart_S\Documents\Arduino\libraries\GirsLib\LedLcdManager.h:7:99: fatal error: LiquidCrystal_I2C.h: No such file or directory
#include <LiquidCrystal_I2C.h> //https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library

So I added Arduino-LiquidCrystal-I2C-library to my Arduino IDE and was able to compile further.

[2] GirsLib/LedLcdManager.cpp generates:
C:\Users\Bart_S\Documents\Arduino\libraries\GirsLib\LedLcdManager.cpp: In static member function 'static void LedLcdManager::setupLcdI2c(int8_t, uint8_t, uint8_t)':
C:\Users\Bart_S\Documents\Arduino\libraries\GirsLib\LedLcdManager.cpp:98:14: error: 'class LiquidCrystal_I2C' has no member named 'init'
         lcd->init();

So I commented out that line as well and I was finally able to compile GirsLite!

bengtmartensson, 2016-01-11:
Ok, here are two problems: 
1. If the CPP symbol LCD is not defined, then it _should_ compile without sucking in that stuff, but obviously, unfortunately I goofed up. Sorry.  So let's accept my goof-up and get it working WITH it.
2. There is definitely an init() in my LiquidCrystal_I2C 

    name=LiquidCrystal I2C
    version=1.1.2
    author=Frank de Brabander
    maintainer=Marco Schwartz <marcolivier.schwartz@gmail.com>
    sentence=A library for I2C LCD displays.
    paragraph= The library allows to control I2C displays with functions extremely similar to               LiquidCrystal library. THIS LIBRARY MIGHT NOT BE COMPATIBLE WITH EXISTING SKETCHES.
    category=Display
    url=https://github.com/marcoschwartz/LiquidCrystal_I2C
    architectures=avr

Do you have another version?

BartmanEH, 2016-01-11:
I'm using the version required by Girs.cpp which is:
#include //https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library

bengtmartensson, 2016-01-11:
Sorry for that.The comment is wrong, will change it. The parameters in the previous message are correct, as is RAEDME.md. The library manager will find it too, but then an older version, and it knows two different libs with that name. Sigh...

BartmanEH, 2016-01-11:
OK, downloaded https://github.com/marcoschwartz/LiquidCrystal_I2C and it all compiles nicely now. Later today I'll try playing with IrScrutinizer and we can get back on topic discussing my results with it!  Thanks for your support.

bengtmartensson, 2016-01-11:
Thank _you_ for telling me what is not working so that I can fix it! Unfortunately, most people just either
1. fix the problem for themselves, or
2. give up

In neither case I can learn and fix for others.

BartmanEH, 2016-01-13:
I'm not getting predictable results with my IR receiver so I asked my local electronic store to get any of the following:
TSMP58000
TSMP58138
TSMP1138
TSMP4138
QSE157 (end-of-life)
OPL551
QSE159
Honeywell SDP8610

They said they will order me a QSE157 (the end-of-life one!). Once that comes in, I'll try again.

bengtmartensson, 2016-01-13:
Great. (IIRC OPL551 is the official successor  of QSE157) Note that the pinout on the data sheet, while probably not ambiguous for a ME (which I am not), can be misunderstood: I fried my first one...

BartmanEH, 2016-01-13:
Thanks for the heads up. Maybe I shouldn't connect it directly to three digital I/O pins as shown in your hardware guide for the Nano? I'll use flying leads to connect to +5V, GND and D I/O Pin 8 instead.

bengtmartensson, 2016-01-14:
Then good luck. You can adapt the hardware configuration file if needed or wanted. The only thing fixed (on a atmega328) is pin 3 as output an pin 8 as input for the non-demodulating sensor.

BartmanEH, 2016-01-14:
How did you burn out your first one if you wired it directly into the Nano I/O pins?

Here's how I plan to wire up the sensor:
[image]
This photo shows the demodulating IR Sensor wired up using +5V and GND pins separately from the Digital I/O pin 8.

Looks ok?

bengtmartensson, 2016-01-14:
>How did you burn out your first one i

On a breadboard. Interesting smell....

I am not able to validate the correctness of your circuit. But it looks like an IR LED to me... a QSE15* looks different.

BartmanEH, 2016-01-14:
The photo shows my [phototransistor](http://optoelectronics.liteon.com/upload/download/DS-50-92-0068/S_110_R3208E.pdf) connected with collector to 5V and emitter to 5k resistor to ground.
I'm still waiting for my QSE157 to come in but intend to wire it similarly - using hook up jumpers from its pins to the Uno/Nano. I don't trust using the I/O pins for power and ground to the QSE157 since I can't trace the C code well enough to be sure that the I/O pins are initialized reliably - there's a chance using I/O pins for power/ground that the polarity from the QSE157 perspective might be reversed, even just briefly, during Arduino reset, power up, initialization, etc. that's why I think wiring it up this way should be safer.

BartmanEH, 2016-01-21:
I got a QSE157 this morning and carefully followed the pin out and wired it up: <a href="https://photos.google.com/photo/AF1QipPYdFadCgkEsMUXaW9FJgkazzLUgevV1x2ao0nC" target="_blank">QSE157 photo</a> but I'm not getting anything in IrScrutinizer :-( Here's <a href="https://photos.google.com/photo/AF1QipNi-tdIzJPqY3lVAFidKF85PoaPYxN9Q23sPfxS" target="_blank">another angle</a>.

bengtmartensson, 2016-01-21:
your URLs are not usable ("Error 404")

BartmanEH, 2016-01-21:
Ugh - Google's new abysmal photo service... let me try again:
QSE157 wired it up: <a href="https://goo.gl/photos/LJYWnyoCZM9fNSSy5" target="_blank">QSE157 photo</a>
Here's <a href="https://goo.gl/photos/xW2DHJxFoiaZHSYy7" target="_blank">another angle</a>.

bengtmartensson, 2016-01-21:
looks ok. Please follow [these instructions](http://harctoolbox.org/arduino_nano_part2.html#Testing+the+firmware).

BartmanEH, 2016-01-21:
I used PuTTY on COM9 @ 115200. Here's the results of the 'v' command, then 'm', then 'l':
GirsLite 2015-12-01
Base Transmit Capture Receive Led
ERROR
and finally 'r' which locks up the terminal and it becomes unresponsive.

bengtmartensson, 2016-01-21:
> then 'l' (for led):

that command it takes an argument,therefore the ERROR

> 'r' locks up the terminal and it becomes unresponsive despite firing and IR remote at it.

It is 'r' for the demodulating receiver, that you do not have. So "become unresponsible" is ok. The command for capturing with the non-demudulating reciver is 'a' (analyze).

BartmanEH, 2016-01-21:
Using the 'a' command and firing known IR signal at it I get a '.' response after about 10 seconds. When I say "known IR signal" I mean one that produces predictable results with the IR library 'IRremote'

bengtmartensson, 2016-01-22:
> Using the 'a' command and firing known IR signal at it I get a '.' response after about 10 seconds. 

There is no signal received at pin8.

BartmanEH, 2016-01-22:
Electronics store ordered me the wrong part - apparently I have a QSE159 with an open collector output. I put a pull up resistor on the output and it's now working perfectly. I'm even getting good results in IrScrutinizer. Thanks so much for your help.

Now the real work begins... efforts leading up to this point was just to get IrScrutinizer working so I can try to decode a pesky remote control. I'm sure I'll be soliciting your help with that once I get some useful results.  Thanks again!

bengtmartensson, 2016-01-22:
>I put a pull up resistor on the output 

You can also do that in software, even within config.h. Just put

    #define IRSENSOR_1_PIN_PULLUP

there :-). 

Nice that it works.

BartmanEH, 2016-01-22:
IrScrutizer v1.1.3 is not stable in my Windows 10 environment:

[1] If I use "Scrutinize signal" and alternate between "Capture" and "Scrutinize", after 2-3 times, it locks up and I have to use Task Manager to kill it and relaunch it.

[2] If I use "Scrutinize remote" and hit "Capture", I get about 10 or more good results and then no more. Hitting Capture again, twice, does nothing. I try "Capturing hw" and hitting Open twice but no more results will show up in Scrutinize remote.

I'm still getting results to work with but I need to manually kill and relaunch IrScrutinizer a lot.

bengtmartensson, 2016-01-22:
Please open  a new  ticket.

closed 2016-01-22 by bengtmartensson

BartmanEH, 2016-01-24:
The challenge with me manually adding lines into config.h is that they will get overwritten in the future by updates that you do. Ideally all such options would pre-exist in config.h etc as comments that could simply be uncommented when needed.

i can understand that there are practical limits to how many commented parameters you might endeavour to put into your code merely to support obscure and unlikely configurations.