bengtmartensson/IrScrutinizer #93: Unicode Support for Import possible?

mxbr, 2016-01-12:
Hi, me again ;-) i noticed that the import (whatever type) does not correctly recognize UTF-8 

Example file (Import via Text, separated by comma)
```
←,533 1706 511 597 511 597 490 618 469 639 469 1706 533 1685 511 597 490 1706 511 1706 511 597 490 618 469 618 469 1727 511 1706 511 597 469 1727 511 597 469 618 469 618 469 639 447 618 469 1727 469 618 469 639 447 1727 490 1727 490 1727 490 1727 490 1727 490 618 469 1727 490
ok,8895 4543 490 1706 511 618 469 639 447 639 447 639 469 1727 490 1706 511 618 469 1727 490 1706 511 618 469 639 447 639 447 1706 511 1727 490 618 469 639 447 1727 490 618 469 639 447 639 447 639 447 1727 490 618 469 1706 511 639 447 1706 511 1706 511 1706 511 1706 511 618 469 1706 511
→,170 5930 106 447 149 362 191 319 213 277 277 255 5098 1685 511 597 490 597 469 618 469 618 469 1727 511 1706 511 597 490 1727 511 1706 511 575 490 639 447 725 383 1727 575 1642 575 533 490 1706 575 1642 575 554 490 618 426 746 127 895 426 1727 554 554 447 682 447 703 383 1727 533 1663 533 1685 533 1685 533 554 511 1706 533

```

→  becomes "?" after import. However if i type it manually after the import it works.

With Export (at least to lirc ) again the same, "→" becomes "?".

So it would be nice having full consistent unicode support.

bengtmartensson, 2016-01-12:
IrScrutinizer _is_ consistent: only US_ASCII is supported on text imports and exports  :-).
Other formats behave differently: Girr has no problem importing whatever charset is declared in the heading, and exports UTF-8. 

The only problem on importing non-ascii is that the user must select the charset for the input. (creating yet another parameter). Likewise with exporting text files. Feel free to send a pull request or patch.

Lirc: AFAIK, Lirc has never defined what are allowed names or not. Alec Leamas is of the opinion that only names from /usr/include/linux/input.h  should be used, and these are C identifiers.

mxbr, 2016-01-12:
Uh, that sucks probably ill just revert to placeholders for importing and exporting stuff and convert it back to unicode with an external program then. 

But if you ask me it makes no sense why text import/export cant do unicode, but inside the program you can type unicode. (Was importing from clipboard btw)

[image]

Ultimately i think moving to Unicode wouldnt be that bad, since it doesnt really come with huge disadvantages, but only advantages, like less internationalisation struggle..

I will try to see if i can patch it, but i dont have the build env for irscrutinizer yet unfortunately

bengtmartensson, 2016-01-12:
Unconditionally interpreting a text file as UTF-8 is a no-go. It may make Windows and Latin-1 (ISO 8859-1) very puzzled.

> But if you ask me it makes no sense why text import/export cant do unicode, but inside the program you can type unicode.

Is a B/W scanner and a B/W printer on a computer that can handle color nonsensical?

But the real reason why non-ascii characterset import and export is not implemented is that my capacity is limited. Help welcome.

bengtmartensson, 2016-01-13:
Plan:

* create branch
* importCharsetName and exportCharsetName as properties, accessible from the GUI
* use that for all reads in imports (except Girr (taken from XML decl), binary formats)
* use that for all writes in exports (except binary formats), including XML (Girr)
* make sure that XML xslt-transformations also generates it <xsl:output method="..." encoding="..." />
* add documentation to the manual
* testing

NOT to be done: Support Microsoft's silly BOM.

By using the comparatively difficult "Type the name of the character set" interface complete beginners are discouraged. This is a Good Thing.

bengtmartensson, 2016-01-13:
First shot checked in, in the branch unicode-import-export. Please test. Build available in as "release ci-buld". Thanx.

bengtmartensson, 2016-01-23:
Still waiting for your feedback and test. Use 
https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build-unicode-import-export
if you do not want to build it yourself.

mxbr, 2016-01-24:
Hi Thanks for the addition of the feature!

So i tested your snapshot and it worked partly. Import works fine but the unicode character is replaced by an underscore when exporting to LIRC.

bengtmartensson, 2016-01-24:
Please upload the export.  Note that you have to select  options -> Export options -> character set to _really_ export URF-8.

bengtmartensson, 2016-02-14:
@mxbr:
Still waiting... I suspect the problem is a usage poblem, but I want to be sure.

bengtmartensson, 2016-02-19:
In the absence of a response, I have no other choice than to assume that mxbr is mistaken.  I shall do so.

closed 2016-02-19 by bengtmartensson