bengtmartensson/IrScrutinizer #96: IrScrutinizer v1.1.3 becomes unresponsive and must be killed with Task Manager

BartmanEH, 2016-01-22:
IrScrutizer v1.1.3 is not stable in my Windows 10 environment:

[1] If I use "Scrutinize signal" and alternate between "Capture" and "Scrutinize", after 2-3 times, it locks up and I have to use Task Manager to kill it and relaunch it.

[2] If I use "Scrutinize remote" and hit "Capture", I get about 10 or more good results and then no more. Hitting Capture again, twice, does nothing. I try "Capturing hw" and hitting Open twice but no more results will show up in Scrutinize remote.

I'm still getting results to work with but I need to manually kill and relaunch IrScrutinizer a lot.

bengtmartensson, 2016-01-22:
1. Is this is sensible thing to do? Cannot reproduce on my Win10 machine.
2. Cannot reproduce. Just captured > 100 signals without any glitches.

Possibly that is a driver problem, or a problem with Java.

Cf. #20.  Will close, nothing I can do. Constructive suggestions welcome.

bengtmartensson, 2016-01-23:
In all cases, please state more details, your Windows version (32/64 bit?, build version), memory, Java version, driver used including version. hardware used (ok, I think I know that :-)) Is there something _reproducible_ experiment that leads to a hanger?

BartmanEH, 2016-01-23:
Windows 10 Pro 64-bit v1511
Java 1.8.0_66
8GB Ram
Driver? driver for what?
[1] and [2] above are reproducible

but it's all in vain. the remote I want to decode produces the following results with IrScrutinizer:
LocalSerialPortBuffered.sendString: Sent '
'.
LocalSerialPortBuffered.readString: received "GirsLite 2015-12-01"
LocalSerialPortBuffered.sendString: Sent '
'.
LocalSerialPortBuffered.readString: received "OK"
LocalSerialPortBuffered.sendString: Sent 'version
'.
LocalSerialPortBuffered.readString: received "GirsLite 2015-12-01"
version returned 'GirsLite 2015-12-01'.
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +139 -308 +139 -308 +139 -643 +140 -475 +140 -643 +139 -307 +140 -307 +140 -811 +140 -307 +139 -24384 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -308 +139 -308 +139 -811 +139 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +139 -476 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +139 -476 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -308 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +140 -307 +140 -308 +139 -811 +140 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +139 -308 +139 -308 +140 -811 +139 -308 +139 -24400 +391 -308 +140 -307 +140 -643 +139 -476 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -308 +139 -24416 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -308 +139 -308 +139 -643 +140 -475 +140 -475 +139 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +392 -307 +140 -308 +139 -643 +140 -475 +140 -476 +139 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -307 +140 -643 +139 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24048 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -643 +139 -643 +140 -475 +140 -643 +140 -308 +139 -308 +140 -811 +140 -308 +140 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -307 +140 -307 +140 -643 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24048 +391 -308 +140 -307 +140 -643 +140 -475 +139 -475 +140 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -308 +140 -307 +140 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +139 -308 +139 -643 +140 -475 +140 -475 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -308 +139 -308 +139 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -308 +139 -308 +139 -643 +139 -476 +139 -475 +140 -308 +139 -308 +139 -308 +139 -308 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +139 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -308 +139 -644 +139 -476 +139 -643 +140 -308 +139 -308 +140 -811 +140 -308 +139 -24384 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +139 -30 [... cut ...] 43 +140 -307 +140 -307 +139 -811 +140 -307 +139 -24352 +391 -307 +140 -307 +139 -643 +139 -475 +139 -475 +139 -307 +140 -307 +139 -307 +140 -307 +139 -307 +140 -642 +140 -475 +140 -642 +140 -307 +139 -307 +140 -810 +140 -307 +139 -24352 +391 -307 +140 -307 +139 -643 +139 -475 +139 -475 +140 -307 +139 -307 +140 -307 +140 -307 +139 -307 +140 -643 +140 -475 +140 -643 +139 -307 +140 -307 +140 -811 +139 -307 +140 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.

BartmanEH, 2016-01-24:
LocalSerialPortBuffered.readString: received "."
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +391 -308 +139 -308 +140 -643 +139 -475 +140 -476 +139 -308 +139 -308 +140 -307 +140 -308 +139 -644 +139 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +140 -308 +139 -24096 +391 -308 +140 -307 +140 -643 +140 -476 +139 -476 +140 -307 +140 -308 +140 -307 +140 -308 +139 -644 +140 -643 +140 -475 +140 -643 +140 -308 +140 -307 +140 -811 +140 -308 +140 -24112 +391 -308 +140 -307 +140 -643 +140 -476 +139 -476 +140 -307 +140 -308 +139 -308 +140 -308 +139 -644 +139 -644 +139 -476 +140 -643 +140 -307 +140 -308 +139 -811 +140 -308 +140 -24080 +391 -308 +140 -307 +140 -643 +140 -475 +140 -476 +139 -308 +140 -307 +140 -308 +140 -307 +140 -643 +140 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24080 +392 -307 +140 -308 +140 -643 +139 -476 +139 -476 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24080 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24096 +391 -308 +139 -308 +140 -643 +140 -476 +139 -476 +140 -307 +140 -308 +140 -307 +140 -308 +140 -643 +139 -644 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +140 -308 +139 -24096 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +140 -307 +140 -308 +140 -643 +139 -644 +139 -476 +140 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24096 +391 -308 +140 -307 +140 -644 +139 -476 +140 -475 +140 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +391 -308 +140 -307 +140 -643 +140 -476 +139 -476 +139 -308 +140 -307 +140 -308 +139 -308 +140 -643 +140 -643 +140 -475 +140 -643 +140 -308 +139 -308 +139 -811 +140 -307 +140 -24080 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +140 -811 +140 -307 +140 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "."
LocalSerialPortBuffered.sendString: Sent 'analyze
'.

bengtmartensson, 2016-01-24:
> Driver? driver for what?

Windows uses a device  driver for accessing the hardware, in your case a (clone?) Arduino Uno. See the device manager. This is a potential problem source.

Your signal are all monster signals (= very long). Does the problem occur with "standard" signals?

To help locate the problem, I suggest that you add the debug LEDs, at least the "red" and "yellow".  Also, due to your long cables to the QSE, I would suggest soldering (!) a 100 nF capacitor between the Vcc and GND pin of the QSE.

BartmanEH, 2016-01-24:
To preserve the integrity of the forum, I will stick to the lock up problem of IrScrutiner in this thread and start a new thread on the monster IR signal I'm trying to decode.

For drivers, my OSEPP Uno R3 Plus uses FTDI v2.12.0.0

The lockup problem does occur with known remotes - several NEC based ones and Sony ones. In these cases, the Scrutinize remote->Parametric remote table fills in very reliably with expected results but will eventually no longer respond to further IR signals. I can move on to other windows and even see results in the Scrutinize signal tab but returning to the Scrutinize remote tab and even toggling the Capture button will not result in getting further data there.

In terms of hardware hookup, I could not get the software defined pullup to work. Firstly, the challenge with me manually adding lines into config.h is that they will get overwritten in the future by updates that you do. Ideally all such options would pre-exist in config.h etc as comments that could simply be uncommented when needed. i can understand that there are practical limits to how many commented parameters you might endeavour to put into your code merely to support obscure and unlikely configurations.

I tried adding #define IRSENSOR_1_PIN_PULLUP in various locations of config.h but it didn't work and I ended up putting the physical pull up resistor back in circuit. I would happily wire it up per your hardware guide with effectively short leads by inserting the sensor directly into the IO pins. With the middle pin of my QSE159 on pin 8 of the Uno I face a problem - which way do I face the IR sensor to ensure that the adjacent IO pins will apply power to the IR sensor correctly without reverse biasing it and causing an awful smell as I've heard happen to other very savvy people I know :-) If I face the QSE159 outward of the Uno, I would need pin 7 to be GND and pin 9 to be VCC. I cannot find documentation on how to control this.

If I could be assured of IO pin levels properly biasing my QSE159 AND ensure that the pullup works on pin 8, I would have the best hardware wiring configuration.

bengtmartensson, 2016-01-24:
Sorry for misleading you, the pullup is not implemented, https://github.com/bengtmartensson/AGirs/issues/15. 

> but will eventually no longer respond to further IR signals. 

The debug LEDS would tell the state of the Arduino sketch here...

BartmanEH, 2016-01-24:
OK, I found the information about the <a href="http://www.harctoolbox.org/arduino_nano.html#LEDs+%28for+visible+light%29" target="_blank">debug LEDs</a>: it will take some time for me to implement this.

BartmanEH, 2016-01-24:
It's relatively easy to implement (once I get the 5V LEDs or bother with current limiting resistors) to wire up the debug LEDs on my Uno which has discreet pins for almost all IO pins. However, the challenge is more substantial for the clone Pro Micro where the <a href="https://goo.gl/photos/X26gwFwBZbYodkMT8" target="_blank">Pro Micro IO pins are shared</a>. In this case, I've had difficulty properly allocating pins and I'm not sure how to ensure that A6 & A7 are properly assigned to pins 4 and 6 of the Pro Micro board and then reworking GirsLite to use A6 & A7 for the 3rd debug LED. I will abandon using a Pro Micro for now and continue with the Uno.

bengtmartensson, 2016-01-24:
A6 & A7 are not used in my circiuts.

BartmanEH, 2016-01-24:
A4 and A5 are not available on the Pro Micro so GirsLite would need to be reworked to use A6 and A7 if we were to go that route.

bengtmartensson, 2016-01-24:
https://github.com/bengtmartensson/AGirs/blob/master/src/GirsLib/girs_pins_micro.h

BartmanEH, 2016-01-25:
Oh cool, thanks. I'm not a crack c oo programmer - quite a noob hacker actually. 

I take it from this section:
'''
#ifdef LED
#define SIGNAL_LED_1 2
//#define SIGNAL_LED_1_GND 3
#define SIGNAL_LED_4 A0
#define SIGNAL_LED_4_GND 15
#define SIGNAL_LED_3 A2
#define SIGNAL_LED_3_GND A1
#define SIGNAL_LED_2 A3
//#define SIGNAL_LED_1_GND 8
#endif
'''
That I would wire up as follows:
LED 1 from pin 2 to ground
LED 2 from A3 to ground
LED 3 from A2 to A1
LED 4 from A0 to pin 15

(1) Correct?

(2) I need to figure out which LED has what colour. 

(3) I need to figure out how to direct GirsLite to use the micro pin config file. I suspect there's a define elsewhere - I'll look for it.

bengtmartensson, 2016-01-26:
This is turning off-topic for an issue tracker.

closed 2016-01-26 by bengtmartensson