bengtmartensson/IrScrutinizer #97: Decoding a monster IR signal for Bell Fibe TV remote (formerly the venerable Motorola Home and presently Arris)

BartmanEH, 2016-01-24:
I've embarked on a cute little project that requires me to detect the power toggle IR signal from a Bell Fibe TV remote control. This is known to be a problematic signal and a few forums have discussed the problem of using IR extenders and learning remotes with this signal:
<a href="http://www.avsforum.com/forum/93-remote-control-area/1517341-theater-master-mx-500-remote-won-t-learn-bell-fibe-remote-any-help.html" target="_blank">Theater Master MX-500 remote won't learn from Bell Fibe Remote</a>
<a href="http://www.remotecentral.com/cgi-bin/mboard/rc-custom/thread.cgi?33379" target="_blank">Bell Fibe boxes only work with certain IR repeaters, some test results</a>

Hoping to peak Bengt's professional curiosity, I'm embarking on soliciting his excellent help to see if this wild beast can be tamed.  After overcoming a plethora of issues compiling and loading GirsLite (thanks for your help, Bengt!) and <a href="https://github.com/bengtmartensson/harctoolboxbundle/issues/92" target="_blank">various issues getting IrScrutinizer to run in my Windows 10 environment</a>, I am now able to get some results for this IR signal despite some continued <a href="https://github.com/bengtmartensson/harctoolboxbundle/issues/96" target="_blank">stability issues in my environment</a>.

I'm still learning how to use IrScrutinizer, a very comprehensive tool. I'm going through the help provided in the application and perusing the website including this excellent <a href="http://www.hifi-remote.com/wiki/index.php?title=IrScrutinizer_Guide" target="_blank">getting started guide</a>.

Here then are my preliminary results:
LocalSerialPortBuffered.sendString: Sent '
'.
LocalSerialPortBuffered.readString: received "GirsLite 2015-12-01"
LocalSerialPortBuffered.sendString: Sent '
'.
LocalSerialPortBuffered.readString: received "OK"
LocalSerialPortBuffered.sendString: Sent 'version
'.
LocalSerialPortBuffered.readString: received "GirsLite 2015-12-01"
version returned 'GirsLite 2015-12-01'.
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +139 -308 +139 -308 +139 -643 +140 -475 +140 -643 +139 -307 +140 -307 +140 -811 +140 -307 +139 -24384 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -308 +139 -308 +139 -811 +139 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +139 -476 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +139 -476 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -308 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +140 -307 +140 -308 +139 -811 +140 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +139 -308 +139 -308 +140 -811 +139 -308 +139 -24400 +391 -308 +140 -307 +140 -643 +139 -476 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -308 +139 -24416 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -308 +139 -308 +139 -643 +140 -475 +140 -475 +139 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +392 -307 +140 -308 +139 -643 +140 -475 +140 -476 +139 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -307 +140 -643 +139 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24048 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -643 +139 -643 +140 -475 +140 -643 +140 -308 +139 -308 +140 -811 +140 -308 +140 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -307 +140 -307 +140 -643 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24048 +391 -308 +140 -307 +140 -643 +140 -475 +139 -475 +140 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -308 +140 -307 +140 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +139 -308 +139 -643 +140 -475 +140 -475 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -308 +139 -308 +139 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24048 +391 -308 +139 -308 +139 -643 +139 -476 +139 -475 +140 -308 +139 -308 +139 -308 +139 -308 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +139 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +140 -307 +140 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -308 +139 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -308 +139 -644 +139 -476 +139 -643 +140 -308 +139 -308 +140 -811 +140 -308 +139 -24384 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +139 -308 +139 -643 +140 -475 +139 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24384 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24384 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24384 +391 -307 +139 -308 +139 -643 +139 -475 +140 -475 +140 -307 +140 -307 +139 -308 +139 -307 +140 -307 +139 -643 +140 -475 +139 -643 +139 -308 +140 -307 +140 -811 +139 -308 +139 -24384 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24384 +391 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -307 +140 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +139 -24384 +391 -308 +139 -308 +139 -643 +139 -475 +140 -475 +139 -307 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +139 -308 +139 -308 +139 -811 +139 -308 +139 -24368 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -307 +140 -307 +140 -307 +140 -307 +139 -308 +139 -643 +139 -475 +140 -643 +139 -307 +140 -307 +140 -811 +139 -307 +140 -24384 +391 -307 +140 -307 +139 -643 +139  [... cut ...] 75 +140 -307 +140 -307 +140 -308 +139 -308 +140 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +140 -811 +140 -307 +140 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "."
LocalSerialPortBuffered.sendString: Sent 'analyze
'.

bengtmartensson, 2016-01-24:
Quick question: there is a hard limit of signal length 255 in GirsLite (number of durations); do you hit that limit?

BartmanEH, 2016-01-24:
Gonna need some help determining if I'm hitting the hard limit of 255 length - where would I look to see that?

bengtmartensson, 2016-01-24:
It is the "number of numbers" in a signal.

BartmanEH, 2016-01-24:
Well... let's count them:
using Notepad++, the following text reports 302 words:
"f=35714 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +139 -308 +139 -308 +139 -307 +140 -307 +140 -307 +139 -643 +139 -475 +139 -643 +140 -307 +140 -307 +139 -811 +140 -307 +139 -24384 +391 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -308 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +139 -307 +140 -24384 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +140 -307 +140 -308 +139 -308 +140 -307 +140 -307 +140 -643 +139 -476 +139 -643 +140 -307 +140 -308 +139 -811 +140 -307 +140 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -308 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24384 +391 -308 +139 -307 +140 -643 +140 -475 +139 -475 +140 -307 +139 -308 +139 -308 +139 -307 +140 -307 +140 -643 +139 -475 +140 -643 +139 -308 +139 -307 +140 -811 +139 -307 +140 -24368 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -308 +139 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -307 +140 -24368 +391 -307 +139 -307 +140 -643 +140 -475 +139 -475 +139 -308 +139 -308 +139 -308 +139 -307 +140 -307 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +139 -811 +140 -307 +140 -24368 +391 -308 +139 -308 +139 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -307 +140 -308 +139 -643 +139 -475 +140 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24368 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -307"

bengtmartensson, 2016-01-24:
Sorry, there is no magic 255, however there is an DEFAULT_CAPTURESIZE, which is not really a  default in GirsLite. Try to raise DEFAULT_CAPTURESIZE. You may undefine RECEIVE and TRANSMIT to save some RAM.

BartmanEH, 2016-01-24:
I commented out:
//#define TRANSMIT
//#define RECEIVE
and changed:
#define DEFAULT_CAPTURESIZE 400U // must be even

Results for a known NEC1 signal:
LocalSerialPortBuffered.readString: received "f=20000 +0 -7216 +1767 -4543 +529 -607 +529 -580 +555 -581 +529 -607 +529 -581 +555 -581 +529 -607 +529 -581 +555 -1717 +529 -1717 +529 -1717 +555 -1717 +529 -1717 +529 -1717 +555 -1717 +529 -1717 +529 -607 +528 -581 +555 -581 +528 -1717 +555 -581 +529 -607 +528 -1718 +528 -607 +529 -1717 +529 -1717 +555 -1717 +529 -581 +555 -1717 +528 -1718 +528 -607 +529 -1717 +529 -40048 +8986 -2299 +528 -30896 +8986 -2299 +528 -30896 +8986 -2299 +528 -30896 +8986 -2299 +529 -30896 +8986 -2299 +529 -30896 +8986 -2299 +528 -30896 +8986 -2299 +529 -30896 +8986 -2299 +528 -30896 +8986 -2299 +528 -30896 +8986 -2299 +529 -30896 +9012 -2272 +529 -30896 +9012 -2272 +528 -30896 +9013 -2272 +528 -30896 +8986 -2299 +528 -30896 +9013 -2272 +529 -32768"

Results for the monster IR signal:
LocalSerialPortBuffered.readString: received "f=35714 +392 -307 +140 -308 +139 -643 +140 -476 +139 -476 +140 -307 +140 -308 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +140 -644 +139 -308 +140 -307 +140 -811 +140 -308 +140 -24112 +392 -308 +140 -307 +140 -644 +139 -476 +140 -476 +139 -308 +140 -308 +140 -307 +140 -308 +140 -644 +139 -644 +140 -476 +140 -643 +140 -308 +140 -308 +139 -812 +140 -308 +139 -24096 +391 -308 +140 -307 +140 -643 +140 -476 +139 -476 +140 -308 +139 -308 +140 -307 +140 -308 +140 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +140 -308 +140 -24096 +391 -308 +139 -308 +140 -643 +140 -475 +139 -476 +139 -308 +140 -307 +140 -308 +139 -308 +140 -643 +139 -644 +139 -476 +139 -643 +140 -308 +140 -307 +140 -811 +140 -307 +140 -24080 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -308 +139 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -308 +139 -24080 +391 -307 +140 -308 +140 -643 +139 -476 +139 -476 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +140 -476 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +140 -24064 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +140 -307 +140 -643 +139 -475 +140 -475 +140 -308 +139 -308 +139 -308 +140 -307 +140 -643 +139 -643 +140 -475 +140 -643 +140 -308 +139 -308 +139 -811 +140 -307 +140 -24048 +391 -307 +139 -308 +139 -643 +140 -475 +139 -475 +140 -307 +139 -307 +140 -307 +140 -307 +140 -643 +139 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24032 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24064 +392 -307 +140 -308"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +391 -307 +140 -307 +139 -643 +139 -475 +140 -475 +139 -308 +139 -307 +140 -307 +140 -307 +140 -643 +139 -643 +139 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +392 -307 +140 -308 +139 -644 +139 -476 +139 -476 +139 -308 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +140 -475 +140 -643 +140 -308 +139 -308 +140 -811 +140 -308 +139 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +139 -308 +139 -24080 +391 -308 +139 -308 +140 -643 +139 -476 +139 -476 +139 -308 +140 -307 +140 -308 +139 -308 +140 -643 +139 -643 +140 -475 +140 -643 +140 -308 +139 -308 +140 -811 +140 -308 +140 -24096 +391 -308 +140 -307 +140 -643 +140 -476 +139 -476 +140 -307 +140 -308 +139 -308 +140 -308 +139 -643 +140 -643 +140 -476 +139 -644 +139 -308 +140 -307 +140 -811 +140 -308 +139 -24080 +391 -308 +139 -308 +139 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +140 -643 +140 -643 +140 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -308 +139 -24080 +391 -308 +140 -307 +140 -643 +139 -476 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +139 -476 +139 -643 +140 -308 +139 -308 +139 -811 +140 -308 +139 -24080 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -643 +139 -476 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24080 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +140 -307 +140 -307 +140 -308 +139 -643 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +140 -811 +139 -308 +139 -32768"

bengtmartensson, 2016-01-24:
The signal should end with -32768 or something, not -308, so 400 also seems to small.

BartmanEH, 2016-01-24:
I tired 600U several times but GirsLite never seems to return results to IrScrutinizer. Using 500U I get the following:
LocalSerialPortBuffered.readString: received "f=35714 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -308 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +140 -811 +139 -308 +140 -24432 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -308 +139 -644 +139 -476 +139 -476 +139 -308 +140 -307 +140 -307 +140 -308 +139 -308 +139 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -308 +139 -24416 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +139 -476 +139 -643 +140 -308 +140 -307 +140 -811 +140 -307 +140 -24432 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -308 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -476 +139 -643 +140 -308 +140 -307 +140 -811 +140 -307 +140 -24416 +391 -308 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -308 +139 -308 +139 -308 +139 -308 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +140 -307 +140 -24416 +392 -307 +140 -308 +139 -644 +139 -476 +139 -476 +139 -308 +140 -307 +140 -308 +140 -307 +140 -308 +139 -644 +139 -476 +139 -643 +140 -308 +139 -308 +140 -811 +140 -307 +140 -24416 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24400 +391 -308 +140 -307 +140 -643 +139 -476 +139 -476 +139 -308 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +140 -811 +140 -307 +140 -24416 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -32768"
Undecodable signal, ignored
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "."
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "."
LocalSerialPortBuffered.sendString: Sent 'analyze
'.

Which is similar to what was received before.

bengtmartensson, 2016-01-24:
But that is excellent! IrScrutinizer breaks it down to 12 repetitions (with no residue!!) of protocol = Nokia32, device = 37, subdevice = 0, obc = 12, misc = X=38 T=0

BartmanEH, 2016-01-24:
If I give it short bursts, I get results e.g.:
LocalSerialPortBuffered.readString: received "f=35714 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -308 +139 -644 +139 -476 +139 -643 +140 -308 +140 -307 +140 -811 +139 -308 +140 -24416 +391 -307 +140 -308 +139 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +139 -476 +139 -643 +140 -307 +140 -308 +139 -811 +140 -307 +140 -32768"
and
LocalSerialPortBuffered.readString: received "f=35714 +392 -307 +140 -307 +140 -643 +139 -476 +139 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +140 -643 +139 -475 +140 -643 +139 -308 +140 -307 +139 -811 +139 -308 +140 -24048 +391 -307 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -24048 +391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -307 +140 -307 +140 -643 +140 -643 +140 -475 +139 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -32768"
or
LocalSerialPortBuffered.readString: received "f=35714 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -308 +140 -643 +139 -643 +140 -476 +140 -643 +139 -308 +140 -307 +140 -811 +139 -308 +140 -24064 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +139 -643 +140 -475 +140 -643 +140 -307 +140 -308 +140 -811 +140 -307 +140 -32768"

but IrScrutinzer always says :
Undecodable signal, ignored

Where do I see results like you report? e.g. "protocol = Nokia32, device = 37, subdevice = 0, obc = 12, misc = X=38 T=0"?

bengtmartensson, 2016-01-24:
What DecodeIR version does Help->About report?

BartmanEH, 2016-01-24:
DecodeIR version 2.45

bengtmartensson, 2016-01-24:
thats good. What I did was to paste 

    +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -308 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +140 -811 +139 -308 +140 -24432 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -308 +139 -644 +139 -476 +139 -476 +139 -308 +140 -307 +140 -307 +140 -308 +139 -308 +139 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24400 +391 -307 +140 -307 +140 -643 +140 -475 +139 -476 +139 -308 +139 -308 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -308 +139 -24416 +391 -307 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +139 -476 +139 -643 +140 -308 +140 -307 +140 -811 +140 -307 +140 -24432 +391 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -308 +140 -307 +140 -308 +139 -308 +140 -307 +140 -643 +140 -476 +139 -643 +140 -308 +140 -307 +140 -811 +140 -307 +140 -24416 +391 -308 +140 -307 +140 -643 +140 -475 +139 -475 +140 -307 +140 -308 +139 -308 +139 -308 +139 -308 +140 -643 +140 -475 +140 -643 +140 -307 +140 -308 +139 -811 +140 -307 +140 -24416 +392 -307 +140 -308 +139 -644 +139 -476 +139 -476 +139 -308 +140 -307 +140 -308 +140 -307 +140 -308 +139 -644 +139 -476 +139 -643 +140 -308 +139 -308 +140 -811 +140 -307 +140 -24416 +392 -307 +140 -307 +140 -643 +140 -475 +140 -475 +139 -308 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24400 +391 -308 +140 -307 +140 -643 +139 -476 +139 -476 +139 -308 +140 -307 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +140 -811 +140 -307 +140 -24416 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -308 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +140 -24400 +391 -307 +140 -308 +139 -643 +140 -475 +139 -475 +140 -307 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +140 -475 +140 -643 +140 -307 +140 -307 +140 -811 +139 -308 +139 -32768

to the upper window of "Scrutinize signal",  and pressing "Scrutinize"; with and without repeatfinder enabled (Options -> invoke repeat finder),  Does DeccodeIR identify other signals, like NEC?

BartmanEH, 2016-01-24:
DecodeIR is producing NEC1 for another remote I test with. Pasting this:
+8985 -4545 +528 -581 +555 -581 +529 -607 +529 -581 +555 -581 +528 -607 +529 -581 +555 -581 +528 -1717 +555 -1717 +529 -1717 +528 -1717 +555 -1717 +529 -1717 +528 -1717 +555 -1717 +529 -581 +555 -581 +528 -608 +528 -581 +555 -1717 +528 -581 +555 -1717 +528 -581 +555 -1717 +528 -1717 +529 -1717 +555 -1717 +528 -581 +555 -1717 +529 -581 +555 -1717 +529 -40048 +8986 -2272 +555 -30880 +8986 -2272 +555 -30880 +8986 -2272 +555 -32768
Yields: protocol = NEC1, device = 0, obc = 80
but pasting this:
+391 -307 +140 -307 +140 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -643 +140 -475 +139 -643 +140 -307 +140 -307 +140 -811 +140 -307 +140 -24064 +391 -308 +139 -308 +140 -643 +140 -475 +140 -475 +140 -307 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +139 -475 +140 -643 +139 -308 +140 -307 +140 -811 +140 -307 +140 -24064 +392 -307 +140 -308 +139 -643 +140 -475 +140 -475 +140 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +140 -475 +140 -643 +139 -308 +139 -308 +140 -811 +140 -307 +140 -32768
yields blank DecodeIR results

bengtmartensson, 2016-01-24:
By the second one I become
protocol = Nokia32, device = 37, subdevice = 0, obc = 12, misc = X=38 T=1

select options -> debug -> debug code and enter 2048, then repeat the last one.

BartmanEH, 2016-01-24:
just blank for me:
LocalSerialPortBuffered.readString: received "f=36363 +391 -308 +139 -307 +140 -643 +139 -475 +140 -475 +140 -307 +140 -307 +140 -307 +140 -307 +140 -643 +139 -643 +140 -475 +139 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24048 +391 -308 +139 -308 +139 -643 +139 -475 +140 -475 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +139 -643 +139 -475 +139 -643 +139 -308 +139 -308 +139 -811 +140 -307 +140 -24064 +392 -307 +140 -307 +140 -643 +139 -476 +139 -476 +139 -308 +139 -308 +140 -307 +140 -307 +140 -643 +140 -643 +139 -476 +139 -643 +140 -308 +139 -308 +140 -811 +140 -308 +139 -24064 +391 -307 +140 -307 +140 -643 +139 -476 +139 -476 +139 -308 +139 -308 +139 -308 +139 -308 +139 -643 +140 -643 +139 -475 +140 -643 +139 -308 +139 -308 +139 -811 +139 -308 +140 -24080 +391 -308 +140 -307 +140 -643 +140 -475 +140 -475 +140 -307 +140 -308 +139 -308 +139 -308 +140 -643 +139 -643 +140 -475 +140 -643 +139 -308 +139 -308 +140 -811 +140 -307 +140 -32768"
Debug[DecodeIR]: DecodeIR was setup with f=36363, repeat=36, ending=0, data=[391, 308, 139, 308, 140, 643, 139, 475, 139, 475, 139, 308, 139, 308, 139, 308, 139, 308, 140, 643, 140, 643, 139, 475, 140, 643, 139, 308, 139, 308, 139, 811, 139, 308, 140, 24064]

bengtmartensson, 2016-01-24:
Bad news. Your Win64-DecodeIR gives different result than mine (Linux 64bit). There is some genuinely bad programming therein,   (http://www.hifi-remote.com/forums/viewtopic.php?t=14219&postdays=0&postorder=asc&&start=47)  (I am "Barf" there). DecodeIR is not my software, and I intend to replace it (https://github.com/bengtmartensson/IrpTransmogrifier)

BartmanEH, 2016-01-24:
OK. moving on....
I see you've been very active in the IRlib forum. What I would like is a simple way to receive this Nokia32 signal. When I use IRremote, I get nice simple hex code results very reliably. How do I go about getting an easy to use extended version of IRremote that can interpret this Nokia32 signal?

bengtmartensson, 2016-01-24:
Nonono. If you absolutely want to jump to conclusions, let me suggest using Linux instead of Windows, the problem does not occur there. 

Personally, I do not hold IRLib and IRremote in very high regard.  Also, they are libraries, nor programs.  I have made myself a picture of the qualifications of the maintainers, both their knowledge in the problem domain, and as programmers. There is a reason why I did not use these libraries.

DecodeIR is still *by far* the most complete decoding library there is.

Will see if I can reproduce the problems on Windows tomorrow.

BartmanEH, 2016-01-24:
Well I appreciate your help and agree that any solution you come up with is bound to be more reliable... especially since I just tried IRLib and didn't get consistent results. I get consistent results with the NEC1 remote but the Nokia32 one usually produces:
Unknown type received. Ignoring.
Decoded Unknown(0): Value:0 (0 bits)
Raw samples(34): Gap:63966
  Head: m360  s340
0:m160 s290	1:m160 s590		 2:m160 s490	3:m160 s440		 
4:m160 s290	5:m160 s240		 6:m160 s290	7:m210 s1040		 
8:m110 s640	9:m160 s440		 10:m210 s590	11:m160 s790		 
12:m160 s290	13:m210 s690		 14:m260 s190	15:m210
Extent=10810
Mark  min:110	 max:260
Space min:190	 max:1040

and only one time produced:

Decoded RCMM: Value:250026CC (32 bits)
Raw samples(36): Gap:54000
  Head: m410  s290
0:m160 s290	1:m160 s640		 2:m160 s440	3:m160 s440		 
4:m160 s290	5:m210 s240		 6:m160 s290	7:m160 s240		 
8:m210 s290	9:m160 s590		 10:m160 s440	11:m160 s640		 
12:m110 s790	13:m160 s290		 14:m160 s790	15:m210 s240		 

16:m160
Extent=10460
Mark  min:110	 max:210
Space min:240	 max:790

bengtmartensson, 2016-01-26:
I have been able to reproduce the problem on Winddows. No promises....

BartmanEH, 2016-01-26:
I understand. Ultimately I need a library for Arduino and not Windows though. Are we headed that way?

bengtmartensson, 2016-01-26:
Remark; Nokia-32 is not a "monster"  (some airconditioning signals are), only a very large number of repeats makes it a "monster". The idiosyncrasies of that protocol are well understood.

The problem with DecodeIR transferred to #98. IrScrutinizer has (with the exception of the DecodeIr problem) no issues with sending or recognizeing Nokia32. The rest is off-topic for an issue tracker.

bengtmartensson, 2016-01-30:
The quoted signal has somewhat too short marks. New fix: See https://github.com/bengtmartensson/AGirs/issues/17 and https://github.com/bengtmartensson/Infrared4Arduino/issues/6. I.e. get and flash a new GirsLite, available soon.

bengtmartensson, 2016-01-31:
@BartmanEH :
Please try the just committed AGirs + Infrared4Arduino firmware. 
If it still needs tweaking, try IRSENSOR_MARK_EXCESS in config.h

BartmanEH, 2016-01-31:
I've downloaded the new AGirs + Infrared4Arduino and copied the appropriate folders into Arduino IDE library folder but getting a compile error:
```
C:\Users\Bartman\Documents\Arduino\libraries\GirsLib\LedLcdManager.cpp:4:32: fatal error: ../GirsLite/config.h: No such file or directory
 #include "../GirsLite/config.h"
```
I have copied the contents of *Girs.cpp* into *GirsLite.cpp* and I have even selected the Arduino Nano board type since *LedLcdManager.cpp* selects different *config.h* file based on this:
```
#ifdef ARDUINO_AVR_NANO
#include "../GirsLite/config.h"
#else
#include "../Girs/config.h"
#endif
```

bengtmartensson, 2016-01-31:
&%$&(=(=()!!"§%&!!

The Arduino IDE is always ready to hit back on those who tries to escape its twisted logic. For now, you should be able just to delete the quoted five lines; config.h is just used to pick up the symbol LCD, which you do not use anyhow.

Real sorry for that.

BartmanEH, 2016-01-31:
following up on this thread for completeness.... the latest committed AGirs does not require commenting out the lines reference immediately above since Bengt has made some programmatic changes to accommodate the finicky Arduino IDE - well done, Bengt!

BartmanEH, 2016-02-02:
Some experimentation with the newly committed code shows that each press of the power button on this remote control yields a toggling result:
```
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=35714 +401 -298 +149 -298 +150 -633 +150 -465 +150 -465 +149 -298 +150 -297 +150 -298 +149 -298 +149 -633 +150 -633 +149 -465 +150 -633 +149 -298 +150 -297 +150 -801 +150 -298 +149 -24054 +401 -298 +149 -298 +149 -634 +149 -465 +150 -465 +149 -298 +149 -298 +150 -298 +149 -298 +149 -633 +150 -633 +149 -466 +149 -633 +150 -297 +150 -298 +149 -801 +150 -298 +149 -32758"
LocalSerialPortBuffered.sendString: Sent 'analyze
'.
LocalSerialPortBuffered.readString: received "f=36363 +401 -298 +150 -297 +150 -633 +149 -465 +150 -465 +150 -298 +149 -298 +149 -298 +150 -297 +150 -298 +149 -633 +150 -465 +150 -633 +150 -297 +150 -298 +149 -801 +149 -298 +149 -24406 +401 -297 +150 -298 +149 -634 +149 -466 +149 -466 +149 -298 +150 -298 +149 -298 +149 -298 +149 -298 +150 -633 +149 -466 +149 -633 +149 -298 +150 -298 +149 -801 +149 -298 +149 -32758"
```
IrScrutinizer still shows blank IrDecode results.

bengtmartensson, 2016-02-02:
> Some experimentation with the newly committed code shows that each press of the power button on 
> this remote control yields a result that toggles between two results:

Correct, It is [part of the protocol](http://www.harctoolbox.org/Glossary.html#toggle)

> I am also having trouble distinguishing different button codes. The "power" button yields very similar 
> results to the "1" button:

Again, consistent with the theory. See 3FG's answer in the link you originally quoted. My IrScrutinizer/DecodeIR  can distinguish them (copy the signal except the f=xxxx and paste it to "Scrutinizer signal").

> I then tried tweaking:
> IrScrutinizer still shows blank IrDecode results.

weird. Ideally, the smallest number found there should be 164, not 149 (was 140 previously), So please try -25.

BartmanEH, 2016-02-02:
It works! It works! Using *#define IRSENSOR_MARK_EXCESS -25* I get consistent results in IrScrutinizer identifying the button and the toggle...amazing! You are truly a genius at this IR stuff!

Now pardon my ignorance but how do I make this usable in my own Arduino project? Do I use your Infrared4Arduino library? How does the *#define IRSENSOR_MARK_EXCESS -25* make its way into that library?

bengtmartensson, 2016-02-02:
Nice to hear of the success!

> how do I make this usable in my own Arduino project? 

For "deployment", you probably want to use a demodulating receiver, TSOPxxxyy (yy = modulation frequency in kHz). (They are easier to locate...)

> Do I use your Infrared4Arduino library?

Of course :-).   Use an appropriate receiving class, like IrReceiverSampler, possibly with a decoder class like Nec1Decoder.

> How does the #define IRSENSOR_MARK_EXCESS -25 make its way into that library?

Member function `setMarkExcess(int16_t)`

BartmanEH, 2016-02-02:
Thank you for pointing me in the right direction.
> Use an appropriate receiving class, like IrReceiverSampler, possibly with a decoder class like Nec1Decoder [...] Member function setMarkExcess(int16_t)

This will take me some time to figure out since I'm not an experienced C++ developer.

bengtmartensson, 2016-02-03:
You were the one asking for library usage. Configuring AGirs is probably easier, it that does what you want.

closed 2016-02-03 by bengtmartensson

BartmanEH, 2016-02-03:
As an inexperienced programmer, I've found it easy to learn how to use the *IRremote* library for other IR projects that are deployed in self-contained Arduino boards. This requires a simple sketch like this:
```
#include <IRremote.h>
#include <IRremoteInt.h>

//initialize included library tools
IRrecv irrecv(RECV_PIN);    //create instance of 'irrecv'
decode_results results;     //create instance of 'decode_results' to parse binary IR data and convert to hex codes
[...]

void setup() {
	[...]
	irrecv.enableIRIn();      //start the IR receiver
}//void setup()

void loop() {
  if (irrecv.decode(&results)) {            //IR signal?
    //Returns true if a code was received, or false if nothing received yet
    //When a code is received, information is stored into "results"
    //  results.decode_type: Will be one of the following: NEC, SONY, RC5, RC6, or UNKNOWN
    //  results.value: The actual IR code (0 if type is UNKNOWN)
    //  results.bits: The number of bits used by this code
    //  results.rawbuf: An array of IR pulse times
    //  results.rawlen: The number of items stored in the array
    //Serial.println(results.value, HEX);   //debug print IR results
    [...switch statement based on results...]
  }//if
}//void loop()
```
using this library, I can get IR results and drive the rest of my sketch depending on the result with a switch statement.

Now am I researching how I can accomplish this with Infrared4Arduino or AGirs.

BartmanEH, 2016-02-05:
Adapting AGirs is proving to be well beyond my programming skills. I have two impediments:

1. converting AGirs server architecture to simply an IR receiving library
2. maintaining these extensive modifications when AGirs changes in the future

What I need is a new version of IRremote library that supports this Nokia32 RCMM protocol and its idiosyncratic trailing mark timing anomaly. Do you have suggestions on how I could get help to accomplish this? Is there a simple modification that could be done to IRremote to tweak the trailing mark excess value

bengtmartensson, 2016-02-07:
Since I do not know what you try to achieve, I cannot offer a sensible answer.

I do not support IRremote.

BartmanEH, 2016-02-07:
I'm trying to:

1. a stand alone Ardiuno Sketch
2. deterministically decode button presses from this remote
3. effect different actions based on button detection including sending completely different IR codes/protocols

bengtmartensson, 2016-02-07:
Ok, then
1. Use a demodulating decoder like TSOP34438,
2. Use an ordinary NEC1 remote, not a finicky Nokia-32,
3. Use the NEC1-decoder, and
4. Write a sketch on the basis of Opponator, you can leave out the LCD

BartmanEH, 2016-02-07:
As per my very first post:
>I've embarked on a cute little project that requires me to detect the power toggle IR signal from a Bell Fibe TV remote control.
I have already made several successful project using NEC1 protocol remotes. This project requires me to specifically work with this finicky remote. This is the whole reason I've worked so hard to get IrScrutinizer/GirsLite to work - just to figure out what protocol this specific remote uses. It has taken weeks of time and considerable effort to get to the eureka moment of determining that this remote uses Nokia32 RCMM with a tweak to the signal required to successfully decode it. Now I wish to finally make my Arduino Sketch to work with this remote.

BartmanEH, 2016-02-10:
I've looked through some of the examples like Opponator and can generally follow the code and can see the possibility of modifying this myself for my application, however, what is missing is a decoder for this fussy Nokia-32 RCMM remote control since I can't use the Nec1Decoder library.

bengtmartensson, 2016-02-10:
Just wait for [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier), see 3. on the README. There is more written code that is not yet on Github...

BartmanEH, 2016-07-11:
Do you think IrpTransmogrifier could do the job for me of detecting/decoding this IR signal?

bengtmartensson, 2016-07-11:
Yes. It will generate code for decoder to be used in Infrared4Arduino.