bengtmartensson/IrScrutinizer #99: High Resolution / DPI on Windows

sippi90, 2016-01-30:
Hi there could you adjust IrScrutinizer for high Resolution?
At the moment on a 4K Resolution it's not really usable!
[image]

bengtmartensson, 2016-01-30:
I am not sure what point you want to make. The screen shot shows that when the GUI is "ridiculously" expanded, it behaves well -- everything is readable, nothing overlaps, what should go right goes right, etc.

If you request that the GUI should look "optimized for 3840 x 2160" I simply reject that request -- it is not meant for that. Just don't maximize and you are set. 

If you mean that larger user interface elements are needed (font, icons, buttons etc) are needed/wanted due to DPI >> 96 or other reason ("accessability") that might be more sensible. Feel free to submit patches and push requests. To me, there are so many more important things, 

Make sure you are using the very latest Java.

sippi90, 2016-01-30:
Yes please add lager user interface elements

bengtmartensson, 2016-02-01:
for the one wanting to work on this: cf https://github.com/arduino/Arduino/pull/4376

bengtmartensson, 2016-11-21:
Possibly relevant: https://github.com/arduino/Arduino/pull/5556

bengtmartensson, 2019-08-14:
Basically everything (or almost everything) should be scalable and the scaling determined by -- what? I have looked at the Arduino IDE and Firefox, and basically learned that there is no easy and immediate solution.

But I still consider most uses of "maximize" silly, an artifact from MS-DOS...

Anyone who is willing and able to help?

bengtmartensson, 2020-09-04:
https://wiki.archlinux.org/index.php/HiDPI#AWT/Swing seems to be at least a crude, solution.
Also see https://docs.oracle.com/javase/8/docs/technotes/guides/2d/flags.html

bengtmartensson, 2023-12-28:
There is a --scale argument, recently improved (#513), that may be satisfying for the original poster. Using the redhat Java 17 on Fedora, it looks quite good to me. Still. it is not a perfect solution, but that lies within Java/Swing. For that reason, I do not think it is meaningful to keep this issue open.

closed 2023-12-28 by bengtmartensson