bengtmartensson/IrScrutinizer #100: Allow relocation of installed "tree", including file names stored in properties.

bengtmartensson, 2016-02-14:
Possible strategy: if a file is a child of `IRSCRUTINIZERHOME`, strip the leading  `IRSCRUTINIZERHOME` from its path, making it relaitve (to `IRSCRUTINIZERHOME`).

Feature branch: relocation

probonopd, 2016-02-29:
What do I need to do to benefit from the solution? Or does it just work now?

bengtmartensson, 2016-02-29:
Just check out branch relocation, or, better,  appimage.

closed 2016-03-22 by bengtmartensson

probonopd, 2016-03-22:
Thanks. I am currently downloading IrScrutinizer-bin.zip from https://github.com/bengtmartensson/harctoolboxbundle/releases" - this does not profit from the solution?

https://github.com/probonopd/AppImages/blob/master/recipes/irscrutinizer/Recipe

bengtmartensson, 2016-03-22:
I am not sure if I understand your question, or if it is a question? 

The subject of this very issue is solved, period. Although for some reason I have not yet merged the branch into master, will do, soon.

The only remaining issue with IrScrutinizer + AppImage is the Java desktop browsing, more precisely   java.awt.Desktop.browse(URI)j (ava.awt.Desktop.edit(....) appears to be working...). I consider this an incompatibility/bug between AppImage and Java/JVM. For the functionality of IrScrutinizer, it is important enough to be considered a showstopper.

probonopd, 2016-03-22:
> The subject of this very issue is solved, period. Although for some reason I have not yet merged the branch into master, will do, soon.

Great, that answers my question. Thanks. About `java.awt.Desktop.browse(URI) (java.awt.Desktop.edit(....)` I have no clue unfortunately. Is it caused by the fact that we cd to AppImage/usr?

bengtmartensson, 2016-03-23:
... merged into the master.

> I have no clue unfortunately. Is it caused by the fact that we cd to AppImage/usr?)

Probably not. Here are some more "random facts": happens with both file:-URLs and http:-URLs. Happens with both OpenJDK and Oracle's JDK, both with your build and mine. Have only tried with Firefox though.