bengtmartensson/IrScrutinizer #103: (Irplus) Export should support Toggle Bit for formats which have one and Maybe repeat value as well

irplus-remote, 2016-02-22:
Hi, Export to irplus should support Toggle Bit for formats which have one

Example code for RC5 

The two codes (Toggle bit 0 and 1) are separated by a pipe | 

```
<irplus> 
    <device manufacturer="AEG" model="TV1" format="PRONTO_HEX">

        <button label="POWER">0000 0073 0000 000C 0020 0020 0040 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0040 0020 0020 0040 0020 0020 0CC8|0000 0073 0000 000C 0020 0020 0020 0020 0040 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0020 0040 0020 0020 0040 0020 0020 0CC8</button>

     </device>
</irplus>
       
```

In addition to this (don't know how complex this is or if there are known values) it would be useful to add a reasonable repeat value if the protocol requires it (example: Sony Protocol usually needs 3 repeats to work). It can be set by using repeat.

```
<device manufacturer="AEG" model="TV1" format="PRONTO_HEX" repeat="2">
```

Thanks!

bengtmartensson, 2016-02-23:
1. Definitely a reasonable request. Need #104 first though.

2. Hmm, how do you think that the user should input his desire? Something like the format specific parameters that presently exist for Girr, Wave, Sendir(Globalcache), Pronto Classic? There is a problem since irplus is not a built-in protocol but defined in exportformats.xml.

irplus-remote, 2016-02-23:
Hey, maybe as for 2.  if there is no way to set a checkbox - I would just consider the "T" parameter, if it is set to 0 or 1 just export a single code with the set toggle bit and if it is set to "-" or "*" it would export with both values.

bengtmartensson, 2016-02-25:
To 2.: A pretty general solution, not tied to irplus,  would be to allow the user to enter an XPath expression, (that should match elements only), an attribute name, and an attribute value. That attribute will then be inserted into all nodes matching the  XPath expression. In your case, the XPath would be "/irplus/device", attribute name "repeat", and attribute value "2".  But it is some work...

bengtmartensson, 2016-02-25:
Try [this](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build)

bengtmartensson, 2016-02-25:
First part should work now. Will close the ticket unless there are issues. Please open a separate ticket for 2. if that is still desired.

irplus-remote, 2016-02-25:
Hi, thanks it worked! Repeat isnt that crucial as it can be easily set manually :+1:

bengtmartensson, 2016-02-26:
A reasonably elegant user-level solution to 2. would be to define one or more  extra export formats in  exportformats.xml, setting the attribute in the XSLT. (There is already commented out code for setting the columns attribute.) E.g. change from

    <exportformat name="irplus" ...>
        ....
    </exportformats>

to

    <exportformat name="irplus" ...>
        ....
    </exportformats>
    <exportformat name="irplus-repeat-3" ...>
       ....
       <xsl:attribute name="repeat">3</xsl:attribute>
        ....
    </exportformats>

 This will give you the new export format irplus-repeat-3 with the desired properties.

bengtmartensson, 2016-03-22:
Closing this, since the wish for settable parameters (which I still consider valid) is subsumed in #116.

closed 2016-03-22 by bengtmartensson