bengtmartensson/IrScrutinizer #117: iTach Flex firmware upgrade breaks? irsxrutinizer

gregsyoung, 2016-03-31:
using latest (1.1.3 version of irscrutinizer) 
was working correctly with iTach Flex (earlier firmware) 
just upgraded iTach Flex firmware to latest version ( 710-3000-17) from GC web site, using 
the iHelp application - upgrade was successful and 
iTach flex working ok as expected,

however in irscrutinizer  when selecting itach flex in caputuring HW pane, 
it now reports an error against the firmware "error 03" (in both capturing and 
sending HW panes) ... (IP address confirmed as correct) ..... interestingly 
capture is  not working under above circumstance, however sending is working correctly
(although both show the error 03 msg)

bengtmartensson, 2016-03-31:
Please turn on Options -> Verbose, repeat what fails, and post the console output. 

Error 03 means (http://www.globalcache.com/files/releases/flex-16/API-Flex_TCP_1.6.pdf) "invalid 
or missing module and/or connector address"

gregsyoung, 2016-04-01:
hi bengt
thanks for fast response

I reinstalled irscrutinizer & manually installed the itach flex IP address
I still get the error 03 message for both capture and sending hw.
however pleased to report that regardlessof the error it is actually
reciving and sending IR signals - all ok

below a verbose from the lower panel (successful send )

Sending command
sendir,1:3,37,36000,1,1,96,32,16,16,16,16,16,32,16,32,48,32,16,16,16,16,16,16,16,16,16,16,16,16,16,16,32,32,32,32,32,16,16,32,32,32,32,32,16,16,32,16,16,16,16,32,16,16,32,16,16,16,16,32,32,32,16,2509
to GlobalCache (192.168.1.100)
completeir,1:3,37

and a couple screenshots attached of how the error msg shows.
[image: Inline image 1][image: Inline image 2]

[image: Inline image 3]

summary
.. working OK for Tx and Rx ... just has the error msg which doesn't appear
to
hinder correct operation.

regards
greg

gregsyoung, 2016-04-01:
[image]
[image]
[image]

bengtmartensson, 2016-04-01:
Not enough, I want the connection dialogue. Like

    Sending command getdevices to GlobalCache (itachflex) 
    Connecting socket to 192.168.1.73:4998
    <device,0,0 ETHERNET
    <device,1,1 IR
    <endlistdevices
    Sending command getversion,0 to GlobalCache (itachflex)
    <710-3000-14
    710-3000-14

Try to select something else (e.g. Audio port for sending), restart the program, then select you GlobalCache, which should give you that dialogue.

bengtmartensson, 2016-04-01:
I think I know what is going on. Try the snapshot https://github.com/bengtmartensson/harctoolboxbundle/releases

gregsyoung, 2016-04-02:
I downloaded the IrScrutinizer-1.1.4snapshot-app.zip 
and found the executable jar file buried in directory
(\IrScrutinizer-1.1.4snapshot\IrScrutinizer.app\Contents\Resources\Java\repo)

when executing the splash screen comes up but program will not start. (splash screen remains on screen indefinitely ... I have to eventually  stop the associated java process ) 
I also downloaded the earlier 1.1.3.zip version and executed the same jar in same way - which works (V1.1.3)

Q:am I doing something incorrect? in trying to run the program using this method?

given that V1.1.3 works using same method, it look like there is some problem? with V1.1.4 jar file?

bengtmartensson, 2016-04-02:
> I downloaded the IrScrutinizer-1.1.4snapshot-app.zip 

That is the Mac Os "App" file. You appears to be using Windows, so you should use the "generic binary distribution", [IrScrutinizer-bin.zip](https://github.com/bengtmartensson/harctoolboxbundle/releases/download/ci-build/IrScrutinizer-bin.zip)

One day the countinuous build will create WIndows setup.exe files too... (see #40)

gregsyoung, 2016-04-02:
deleted all previous downloads and installed versions

did fresh download of IrScrutinizer-bin.zip under the "ci build"
after unzip the directory structure looks as per below screen snapshot.
(only executable is the jar ?) - and this one is the one that just freezes on the splash screen
(note I also downloaded V1.1.3 bin.zip file - which produces a similar file structure and I can correctly execute the 
same named jar - which of course runs as V1.1.3)

hence im doing identical for the newer build .... but the jar file just hangs on the splash screen.

am I doing something wrong?

gregsyoung, 2016-04-02:
[image]

bengtmartensson, 2016-04-02:
Try [this](http://www.harctoolbox.org/IrScrutinizer.html#I+did+something+funny%2C+and+now+the+program+does+not+startup%2C+with+no%0Avisible+error+messages.)

I also uploaded a [setup.exe](https://github.com/bengtmartensson/harctoolboxbundle/releases/download/ci-build/IrScrutinizer-1.1.4snapshot.exe)

gregsyoung, 2016-04-02:
V 1.1.4 installed and running ok from setup.exe file.
now correctly reporting the firmware version of itach flex 

issue closed
appreciate your very fast response and assistance.
thanks
greg

closed 2016-04-02 by gregsyoung

bengtmartensson, 2016-04-02:
Glad that it works now. Not really nice of GC to change the API in a non-compatible way, in particular since it would have been easy to still accept the old syntax...

For the record: Compatibility with old GC-100 has been checked (with the firmware 2.4-06, and 3.0-12); they accept the new syntax, even thought this is not documented.

gregsyoung, 2016-04-02:
noted and yes agree
GC not playing very nice with developers when APIs changed like this.