bengtmartensson/IrScrutinizer #124: irpmaster decodeir with raw codes doesn't work

maghrab, 2016-05-23:
Hello,

If I try to run the following command it gives a null pointer exception

irpmaster -p --ccf +4495 4619 530 1725 592 1682 563 1660 520 566 611 537 621 517 589 533 576 540 576 1669 537 1699 602 1630 553 598 592 543 589 533 576 546 572 530 595 1679 589 1656 537 1728 592 533 576 527 618 501 585 537 605 533 585 524 576 543 592 533 595 1676 569 1656 579 1689 537 1725 595 1650 605 46718 4475 4550 598 1663 530 1715 504 1744 602 514 579 527 579 530 592 537 533 556 585 1692 576 1689 533 1718 507 598 572 543 572 537 572 533 589 553 579 1643 533

Result: 
Exception in thread "main" java.lang.NullPointerException
	at org.harctoolbox.IrpMaster.IrpMaster.main(IrpMaster.java:545)
	at org.harctoolbox.irscrutinizer.IrScrutinizer.main(IrScrutinizer.java:112)

bengtmartensson, 2016-05-23:
Thanx for the report.

The problem is that the signal has an odd number of durations, and that is a no-go in the IrpMaster universe. Try adding a dummy number at the end -- physically that means a period of silence at the end of the sequence, then it will work.

Of course, the behavior you describe is not acceptable, and will be fixed.

bengtmartensson, 2016-06-01:
Fixed by 7664cc8. Better user feedback would still be desirable, however.

closed 2016-06-01 by bengtmartensson