bengtmartensson/IrScrutinizer #125: irpmaster command to export lirc files

maghrab, 2016-06-01:
Is there a command to export raw codes to a lirc file similar to irscrutinizer?

bengtmartensson, 2016-06-01:
Short answer: no.

A better command line interface is planned, #26. Since the API is open, it should not be too hard to use that for achieving what you want to do.

maghrab, 2016-06-01:
Thanks!

It seems that the hexadecimal code I am looking for comes from using --analyze flag, however, when I use it on raw codes it always outputs undetermined

maghrab, 2016-06-01:
The hexadecimal code I am looking for comes from analyzeIR
An example would be:
AnalyzeIR: {38.0k,552,msb}<1,-1|1,-3>(16,-8,A:32,1,^112m,(16,-4,1,^110m)+){A=**0x20df40bf**}

When I capture a raw code, it always outputs undetermined

bengtmartensson, 2016-06-01:
Then the raw code is possibly erroneous...my crystal ball is temporarily unavailable...

Have a look at the Arduino/IRremote export format, may be what you are looking for,

maghrab, 2016-06-01:
Can you give me more information about that?

bengtmartensson, 2016-06-02:
http://harctoolbox.org/IrScrutinizer.html#Arduino%2FInfrared4Arduino ?

maghrab, 2016-06-02:
Thanks! Actually, I am looking for a command in terminal ( probably using irpmaster ) to get the hexadecimal keycode such as above. If you export using the lirc format, it will be in there.

bengtmartensson, 2016-06-15:
Any more comments before closing?

maghrab, 2016-06-16:
No, thanks a lot for your help! Great tool, Keep it up.

closed 2016-06-16 by maghrab