bengtmartensson/IrScrutinizer #130: Direct import of online database for IRDB

ChrisVITQ, 2016-07-06:
Hello Bengt,

is it possible to implement a feature that allows you to  import the Girr online database directly?

The database:
https://github.com/probonopd/irdb

I think it would make things way easier to handle if it would be possible to copy the link of a folder (could also be a folder on the local client) and the included codes get available to choose in the IRScrutinizer window.

Thank you and all the best,
Chris

bengtmartensson, 2016-07-07:
1. Girrlib is "My library of IR commands in Girr format", and not am  "internet resource". I do not have any plans in that direction. Nevertheless, I just extended the Girr importer to do directories -- you can just drop a local directory on the import window. 

2. I am not sure if I understand your second request. I think the current IRDB-interface is pretty good as it is; given the limitations of IRDB (sorting per protocol, D, S). Cf #45 . Can you elaborate?

3. Be sure to check out the daily build.

closed 2016-07-13 by bengtmartensson

bengtmartensson, 2016-07-17:
Replaced by #132, for importing _local_ directories.