bengtmartensson/IrScrutinizer #131: AMSR files generated by IRScrutinizer don't work

kylemd, 2016-07-15:
G'day,

Having some issues with the Anymote export function of IRScrutinizer. The attached file was generated out of remote codes that I learnt with a IRToy v2 on IRScrutinizer v1.2.

When attempting to import into the Anymote app it informs me that it is an "Invalid .amsr file".

I'm going to give [ColorTiger's LIRC to AMSR converter](https://www.anymote.io/convert/lirc2amsr.php) a go later tonight to see if that fixes the problem but thought I'd post here first in case it is a known issue.

Cheers!

[anymote_2016-07-15_17-56-08.amsr.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/365497/anymote_2016-07-15_17-56-08.amsr.txt)

bengtmartensson, 2016-07-15:
I tried the current Android version, and for me, your file loads! Looking at it, the only thing that appears questionable is that the ir sequences end with a comma. Try this [exportformats.xml.zip](https://github.com/bengtmartensson/harctoolboxbundle/files/365605/exportformats.xml.zip) (just replace your existing) to see if it helps.

kylemd, 2016-07-17:
Interesting indeed...

I've done the following trying to rule any external factors out...

- Used three different file transfer methods (Pushbullet, USB cable, OTG thumb drive)
- Re-recorded the remote
- Reinstalled Anymote

Still tells me that it is an invalid .amsr file. What version of Anymote do you have installed?

Attached is the file generated with the aforementioned exportformats.xml

[anymote_2016-07-17_13-56-36.amsr.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/367670/anymote_2016-07-17_13-56-36.amsr.txt)
[girr_2016-07-17_13-56-26.girr.txt](https://github.com/bengtmartensson/harctoolboxbundle/files/367669/girr_2016-07-17_13-56-26.girr.txt)

kylemd, 2016-07-17:
Never mind, I found what the issue was. The default Android Marshmallow file picker must have not liked AnyMote and rejected the file when chosen from there. Tried with OI File Manager and works fine.

Sorry for wasting your time! And compliments on the app and support :+1:

closed 2016-07-17 by kylemd

bengtmartensson, 2016-07-17:
Glad that it is sorted out. You should contact Anymote if you want to investigate the issue further. The file is just one monster line (>13000 chars) without a trailing linefeed/carrage return, which may lead to problems with badly written programs.