bengtmartensson/IrScrutinizer #133: "Building from sources" instructions location

HelenFoster, 2016-07-21:
The current readme file says, "Installation, including dependencies, is covered in detail in the appendix of the manual IrScutinizer.html. This is built during the program build. It can also be build directly by the script tools/mk-irscrutinizer-html.sh. It is also directly accessible online (may be behind the version here.)"

This seems circular, and somewhat unfriendly. I'd like to suggest moving that part of the documentation into the readme, or e.g. an INSTALL.md in the same place. This information is only needed for building from source, so it would make sense to keep it with the source, rather than in the user documentation.

bengtmartensson, 2016-07-22:
@HelenFoster : thank you for the suggestion. I do not think it is that bad and definitely not circular; there is both the script and the online version.

I like to keep all relevant information on one "reference manual", instead of spread out in a number of files (like Lirc :-)). Cf #15 -- there is a lot to do...

Anyhow, just checked in some stuff. There a new file BUILD.md is _generated_ from IrScrutinizer.xml, keeping another _product_ in git...

HelenFoster, 2016-07-22:
Thanks. I suggested moving it so it wouldn't be a product, but if keeping all docs in the one place is higher priority, that works too.

closed 2016-07-22 by HelenFoster