bengtmartensson/IrScrutinizer #139: Problem exporting multiple irplus files

HelenFoster, 2016-07-30:
Hi, I can export one irplus file from IrScrutinizer. If I try to export more irplus files, it prints errors in the console and the exported file has 0 bytes. Restarting IrScrutinizer allows one more file to be exported. Using a build of 3d4129c .

```
ERROR:  'Variable 'encoding' is multiply defined in the same scope.'
FATAL ERROR:  'Variable 'creatingTool' is multiply defined in the same scope.'
Variable 'creatingTool' is multiply defined in the same scope.
File /tmp/exports/irplus_2016-07-30_11-22-19.irplus was successfully written.
```

bengtmartensson, 2016-07-30:
Thanx for the bug report. Should work now.

HelenFoster, 2016-07-31:
Thanks, it's working now.

closed 2016-07-31 by HelenFoster