bengtmartensson/IrScrutinizer #140: Exporting both toggle values for Nokia32

HelenFoster, 2016-07-30:
Hi, I see something was implemented for irplus #103 and girr #104 . It works for RC5, if I export the Parametric Remote when the T field is blank. But leaving T blank for the Nokia32 protocol results in an irplus file with empty buttons. If exporting a girr file, there are comments saying "Raw signal requested but could not be generated" and "Pronto Hex requested but could not be generated". Using IrScrutinizer from 3d4129c .

bengtmartensson, 2016-07-30:
The problem here is that, as defined in IrpProtocols.ini Nokia32 has a parameter named "T", but it is not a toggle in the sense of persistence and default value. IrScrutinizer (proper) handles this correctly, when it refuses to render the signal with no value assigned to T. However, I have (in another context) found out that the definition thus is probably questionable (cf  http://www.hifi-remote.com/forums/viewtopic.php?t=13487), and have changed the definition to be a toggling protocol in the normal sense.

HelenFoster, 2016-07-31:
Thanks, it's working.

As another data point, I have a DigiFusion FVRT90 here. The remote that came with it is transmitting Nokia32 with a real toggle bit. Though it turns out the receiver is not picky (unlike my other toggling devices, this one can get to channel 11 without the toggle).

closed 2016-08-01 by HelenFoster