bengtmartensson/IrScrutinizer #141: More parametric LIRC exports

HelenFoster, 2016-07-30:
The manual says, 

> The LIRC-exports are in lircd.conf-format. These uses the raw LIRC format, except for a few well known protocol (presently NEC1 and RC5).

If any more are added at some point, I think it makes sense to prioritise protocols with toggles, since they don't work properly in raw LIRC format (as we recently discussed on the LIRC mailing list).

RECS80 is a good candidate, since it has a toggle and can be encoded as SPACE_ENC in LIRC. Can be worked around because irrecord handles RECS80 perfectly, but would be nice to have.

bengtmartensson, 2016-07-30:
Thank you for the suggestion. You may have noticed that since 1.2, the Lirc export was removed from the Java core, and now sits entirely as XSLT code within exportformats.xml. There it is relatively easy to add new parametrized protocols; nec1 is lines 1620 to 1666, rc5 lines 1669 to 1698. 

Having said that; I do not have any plans presently to improve the Lirc exporter, but I would be happy to accept contributions.

HelenFoster, 2016-07-31:
I'm not quite at the level to do that yet, but might look into it at a later date.

bengtmartensson, 2016-08-02:
The main work is to determine exactly what lircd. conf is to be generated from, e.g., recs80 signals.

HelenFoster, 2016-08-03:
I started working on RECS80, but I think the LIRC SPACE_ENC category covers anything with `<a,-b|c,-d>` (i.e. most of them) and there should be a way to get all of those in one shot.

Looks like most of the rest are Manchester encoded, and I'm not so familiar with how LIRC handles that.

LIRC's RCMM category (2-bit space encoding, which includes Nokia32) doesn't support transmission, so that's not much use.

bengtmartensson, 2016-08-03:
> there should be a way to get all of those in one shot

Hmm, that is awfully close to what I am busy with in IrpTransmogrifier. Stay tuned :-).

> Looks like most of the rest are Manchester encoded, and I'm not so familiar with how LIRC handles that.

That is basically Lirc's RC5 flag.

bengtmartensson, 2016-10-24:
Please test the current version. It supports the protocols 48-nec1 48-nec2 adnotam aiwa akai archer barco blaupunkt bose bryston denon dgtec dishplayer dish_network elan emerson f32 fujitsu fujitsu-56 'gi rg' iodatan jerrold jvc konka logitech matsui metz19 mitsubishi nec1 nec1-rnc nec1-f16 nec2 necx2 pacemss panasonic_old pid-0001 pid-0003 pid-0004 pioneer proton rc5 rc5-7f rc5-7f-57 rc5x rca rca-38 recs80 recs80-0045 recs80-0068 recs80-0090 replay revox samsung20 samsung36 sampo scatl-6 sharp 'sharp{1}' 'sharp{2}' sim2 sony8 sony12 sony15 sony20 streamzap streamzap-57 sunfire tdc-38 tdc-56 thomson thomson7 tivo velleman viewstar x10

It was generated by [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) commit 06d04681fb1cacdb8fd52075c61882fd39d4ff58. Still a few more that can be supported (variables like checksums, variations).

bengtmartensson, 2016-10-31:
Still remains to do: checksum like variables ("Definitions") and such.

bengtmartensson, 2016-11-21:
See [this issue](https://github.com/bengtmartensson/IrpTransmogrifier/issues/6).

bengtmartensson, 2017-02-28:
Many (around half of them) protocols implemented through new lirc.xml.

Open new tickets if there are problems.

closed 2017-02-28 by bengtmartensson

bengtmartensson, 2017-03-23:
@HelenFoster :
> LIRC's RCMM category (2-bit space encoding, which includes Nokia32) doesn't support transmission,

It is now [claimed](https://sourceforge.net/p/lirc/tickets/276/) that it _is_ supported.

commit 285f460b (2017-02-28): New exportformats.d/lirc.xml, generated by IrpTransmogrifier.
Implements most Lirc export of most protocols without definitions
and without expressions in bitfields. Resolves #141.