bengtmartensson/IrScrutinizer #155: LircClient: support named pipe (/var/run/lirc/lircd), if possible

bengtmartensson, 2016-10-23:
... as alternative to the TCP socket. Of course, this is only relevant for systems with named pipes. Despite of this, a JNI-based solution is not acceptable.

bengtmartensson, 2016-10-24:
Correction: /var/run/lirc/lircd is not a named piped, it is a Unix domain socket. There appears to be a consensus that [accessing these in pure Java is impossible](http://stackoverflow.com/questions/170600/unix-socket-implementation-for-java). Linked article contains JNI-projects.

For this reason, rejected.

closed 2016-10-24 by bengtmartensson

leamas, 2016-10-25:
Or, don't implement the unix domain socket but instead install a wrapper  which makes the socket visible as a TCP port. Using  _socat(1)_ it can be a simplle oneliner: like `socat TCP4-LISTEN:9999,fork UNIX:var/lircd.socke` which makes the local socket (test variant) accessible on the localhost:9999 tcp port.

If we should wallk this  way the proper path would be to install it as a systemd service. But for testing, the oneliner might do it.

bengtmartensson, 2016-10-26:
Thanx for telling me about socat; I did not know that before. 

I doubt however that Lirconians will prefer using socat over starting lircd  with the `--listen` option :-). 

If you have further comments, it might be better to write them [here](https://github.com/bengtmartensson/JavaLircClient/issues/1).  (Implementing that task should nor really be major effort,)