bengtmartensson/IrScrutinizer #162: Idea: More insightful graphical representation

probonopd, 2016-12-20:
Probably many users of IrScrutinizer are not just interested in getting the codes to work _per se_ but also want to learn something about their meaning along the way. For this, I would find it useful if the graphical representation could explain a bit more about the decoded protocols, like in the example below from the open source [PulseView](https://sigrok.org/wiki/PulseView) software.

[image]

The above was made with a TSOP attached to a 6 EUR logic analyzer from China and [PulseView.AppImage](https://bintray.com/probono/AppImages/PulseView#files).

Also, I wonder if the protocol decoders (and possibly more) could be shared between the two projects.

bengtmartensson, 2016-12-20:
The idea of annotating/decorating the plot with decoder/IRP information (as well as breaking-up non-interlacing protocols like the bi-phase (Manchester) protocols like RC5, the red row in the picture above) is certainly very attractive. But also quite a lot of work. 

Breaking up non-interlaced protocols is actually pretty much a different task from the IRP interpretation. (Note: Background colors can be used to (space savingly) indicate 0 or 1.)

Note that this should be make in the content of [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier), which will replace IrpMaster, DecodeIR and the Analyzer.

(BTW, I tried PulseView. The program is completely free from non-trivial user documentation. Tried to have it to display a generated CVS. Core dump. Not very impressed...)

Thank you very much for the suggestions!

bengtmartensson, 2016-12-20:
Must wait for IrpTransmogrifer migration, a.k.a. Milestone Transmogrifier.

probonopd, 2016-12-20:
> BTW, I tried PulseView. The program is completely free from non-trivial user documentation.

That is probably because it is "just" the GUI for [Sigrok](https://sigrok.org/wiki/Main_Page) which is well documented imho.

bengtmartensson, 2017-06-07:
It would also be useful and natural to visualize what the repeat finder has done. Possibly an interactive toggle that toggles between repeat-reduced and original signal.

bengtmartensson, 2019-08-14:
"Un-On Hold"-ed" with IrScrutinizer 2.0.1!  I really like the idea, in particular if it can be combined with (interactive) debugging of (non-)decodes. But is yet unclear to me exactly how it can be implemented.

closed 2023-12-25 by bengtmartensson