bengtmartensson/IrScrutinizer #171: Paste short cut (CTRL + v) not working on Windows on latest ci-build

irplus-remote, 2017-02-08:
Hi, previously it was possible to use CTRL + V on scrutinze window. Has this been removed intentionally or is it a bug? Thanks

bengtmartensson, 2017-02-08:
I can confirm that it is not working for me either.

It has neither been removed nor has it explicitly been put there, so I guess that something has happened with Swing/Java. Nevertheless, it makes sense to have it working. 

Workarounds: Right Mouse -> paste or Edit -> Paste signal does it. If you are using X Windows, you can also paste the "selection" with the middle mouse button.

bengtmartensson, 2017-03-02:
Appears to have gone wrong with f2f57773833738bb4dbc228f0818f1ef2e8f6c6b, (fixing #127), added file dropping support, inadvertently breaking pasting support.

bengtmartensson, 2017-03-02:
@irplus-remote : The problem is believed to be fixed. Please check and reopen if appropriate.

Again, many thanx for reporting.

irplus-remote, 2017-03-03:
Thanks, for some reason i can't get the ci-build though. Maybe a temporary issue, will retry later

https://github.com/bengtmartensson/harctoolboxbundle/releases/download/ci-build/IrScrutinizer-bin.zip

bengtmartensson, 2017-03-03:
There has been some issues with travis and/or the Github API. (At least) At this moment, the CI build should be OK though.

irplus-remote, 2017-03-15:
Hi, i've tested and it works partly now. :) However there seems to be a small weakness.

If i use CTRL + A to select all, and CTRL + V to paste a new code, the old input is kept and only attached to it, instead of being deleted (as it would be done usually)

bengtmartensson, 2017-03-15:
Sigh... I "knew" it was not right to reimplement the pasting logic :-\

Thanx for the feedback.

irplus-remote, 2017-03-16:
Also CTRL + C doesnt work :/   (Copy selection)

bengtmartensson, 2017-03-23:
Please try the current CI-build.

bengtmartensson, 2017-04-02:
OK to close?

irplus-remote, 2017-04-03:
I guess so, seems to be ok now

closed 2017-04-03 by irplus-remote

bengtmartensson, 2017-04-03:
Ok, nice to hear that.

Thank you for this (and other!) bug reports; helping to make IrScrutinizer better.

commit 04297911 (2017-03-02): Fixed bug in pasting in "Scrutinize signal" window; resolves #171.
Also new menu entry "Paste" in context menu of same window.