bengtmartensson/IrScrutinizer #172: IRDB and Control Tower database import empty

sle118, 2017-02-09:


sle118, 2017-02-09:
Nevermid...  it says "select me", which I did, and now there are entries.

closed 2017-02-09 by sle118

bengtmartensson, 2017-02-10:
Glad that it is sorted out. 

If you have any suggestions for making it more intuitive, please let me hear.

sle118, 2017-02-10:
Perhaps with the following:
- entries are cached locally for each importer
- the importer sets up an sync job to pull all brands in is constructor
- a simple refresh button to the left of the list to pull again from server

sle118, 2017-02-10:
This would offload the server while allowing smoother GUI interactions as well as quicker operations when using the classes as API. Days aging might also be a good idea

bengtmartensson, 2017-02-11:
Thank you for the suggestions. I found out that it was missing from the manual; fixed.

As I see it, the main problem presently is that it may not be obvious to select "select me to load", as your ticket proves :-).

I do not want to open a connection just because the user selected a pane in a GUI. All of the proposed methods have drawbacks. Fetching stuff in the background can be confusing if the user does not understand what is going on. 

Strictly speaking you are right about "refresh", but the need only arises if the database added a manufacturer since the program was started. This does not happen "very often".