bengtmartensson/IrScrutinizer #191: Protocol Amino-36 (and Amino-56) does not export correctly to lirc (and other formats)

irplus-remote, 2017-03-21:
Hi, Protocol Amino-36 (and Amino-56) does not export correctly to lirc (and other formats)

It just returns empty files

```
begin raw_codes
		name KEY_OK

		name KEY_PAGEUP

		name KEY_PAGEDOWN

		name KEY_A

		name KEY_B
...
```

bengtmartensson, 2017-03-21:
Cannot reproduce; works for me. Please provide more information, preferrably as a Girr export.

PS. Be sure to delete exportformats.xml from versions < 1.4.

irplus-remote, 2017-03-22:
Im using the latest ci build so it should not be possible that residue remained? How can i be sure im using a virgin install?

- I did the following, set the Export format to "LIRC" or "LIRC RAW".
- Paste this code into the Lirc import
```
#Amino_IR_Protocol_Spec_1_02.pdf

begin remote

  name  Amino
  bits           16
  flags RC5|CONST_LENGTH
  eps            10
  aeps          200

  header       1875  1607
  one           268   268
  zero          268   268
  plead         804
  pre_data_bits   16
  pre_data       0xF2FF
#repeat bit:
#  pre_data 0xF6FF
  ignore_mask 0x0400000
  gap          100000
  toggle_bit_mask 0x0
  frequency    36000
#  frequency 56000
  duty_cycle   33

      begin codes
          KEY_OK                   0xE80B
          KEY_PAGEUP               0xDC0E
          KEY_PAGEDOWN             0xDB0D
          KEY_A                    0xE306
          KEY_B                    0xE205
          KEY_C                    0xE508
          KEY_D                    0xE407
          KEY_FAVORITES            0xF408
          KEY_1                    0xFD01
          KEY_2                    0xFC00
          KEY_3                    0xFB0F
          KEY_4                    0xFA0E
          KEY_5                    0xF90D
          KEY_6                    0xF80C
          KEY_7                    0xF70B
          KEY_8                    0xF60A
          KEY_9                    0xF509
          KEY_0                    0xF307
          KEY_FASTFORWARD          0xDE00
          KEY_REWIND               0xE104
          KEY_PLAY                 0xE003
          KEY_PAUSE                0xDF01
          KEY_STOP                 0xDA0C
          KEY_RECORD               0xDD0F
          KEY_MENU                 0xED00
          KEY_INFO                 0xE306
          KEY_EXIT                 0xE306
          KEY_CHANNELUP            0xEE01
          KEY_CHANNELDOWN          0xEB0E
      end codes

end remote

```

Then on scrutize remote you can see that it has been imported

Hit export and check the file - it seems empty :o

bengtmartensson, 2017-03-22:
OK, now I see. The problem is that DecodeIR, which is doing the decodes, calls "Amino-36" what IrScrutinizer proper calls "Amino". Unfortunately, decoding and rendering is not "synchronized". This will change with [IrpTransmogrifier](https://github.com/bengtmartensson/harctoolboxbundle/milestone/6).  We will have to live with it until then.

After having done the import, the correct way is to change the protocol in the remote table to "Amino", For this, use right mouse -> Advanced  -> Set protocol... 

Nevertheless, silently producing a silly export is not acceptable. I made a new ticket for this, #192 .

For this reason, I am closing the ticket.

> How can i be sure im using a virgin install?

`irscrutinizer --nuke-properties`, or just select Options -> Export format database -> Select , and select the _directory_ exportformats.d, _not_ the file exportformats.xml.

closed 2017-03-22 by bengtmartensson