bengtmartensson/IrScrutinizer #193: Open last file does not work anymore

irplus-remote, 2017-03-22:
Hi, i can't use the "Open last file" button anylonger. Any idea?

Error : No application is associated

-> But it is

Maybe related to this? https://github.com/bengtmartensson/harctoolboxbundle/issues/149

bengtmartensson, 2017-03-22:
It is the operating system (which one are you using?) that associates applications and that produces that error message. Which type is failing? Also Java's Desktop can be involved, which Java are you using?

> Maybe related to this? #149

That is with girr only, which are now copied to .txt.

irplus-remote, 2017-03-22:
I'm using Windows, and the files are associated with Notepad++ (any type fails including lirc). Also it worked before. Happening since i am using the latest build. (Java is oracle JDK on windows)

I will now update Notepad++ and Java just to be sure though

bengtmartensson, 2017-03-23:
Please try the CI-build.

irplus-remote, 2017-03-23:
Working again :)

bengtmartensson, 2017-03-23:
Great! Thanx for the bug report.

closed 2017-03-23 by bengtmartensson