bengtmartensson/IrScrutinizer #194: Support for Raspberry Pi GPIO Transceivers

Oliveirakun, 2017-03-23:
Do you have any plans to make scrutinizer run in a raspberry pi using an IR Transceiver connected through GPIO?
There are some GPIO boards for raspberry and in general, they are better to make projects rather than USB boards.

bengtmartensson, 2017-03-23:
It is already supported: Use the lirc_rpi driver ([my version](https://github.com/bengtmartensson/lirc_rpi) is of course the best one :-)), compatible hardware,  and select `/dev/lirc` in IrScrutinizer. Only non-demodulation sensors (i.e. frequency measurement) is not supported.

Having said that,

> There are some GPIO boards for raspberry and in general, they are better to make projects rather than USB boards

Interrupting a processor running Linux every 13 microseconds (for 38k modulation) is not my concept of a technologically sound design ;-).

Oliveirakun, 2017-03-24:
Nice! So, scrutinizer works with this board, right? http://www.irdroid.com/irdroid-rpi-infrared-transceiver/
I'm developing a new project and I'm analysing all the options(boards and software) for raspberry, I'm looking for something better than Lirc and irrecord.
Can I use just IrpMaster? Do you know any other good board to use with raspberry?

bengtmartensson, 2017-03-24:
> scrutinizer works with this board, right? http://www.irdroid.com/irdroid-rpi-infrared-transceiver/

Should work. This is just some trivial components (~ 3€) sold to a juicy price. Of course, "reduced price", probably since the introduction or a few days thereafter...  Still interrupting the CPU every 13 microseconds...

> something better than Lirc and irrecord.

Sure, Do you really need capturing ("irrecord") running on the RPi? Is in not enough to capture on a PC, say? 

> Can I use just IrpMaster?

... to do what?

> Do you know any other good board to use with raspberry?

I like a micro processor to do the timing, possibly also decoding. Possibly [this](http://www.watterott.com/de/RPi-UNO-HAT), a few components (cf [this](http://www.harctoolbox.org/arduino_nano.html))  used with [Agirs](https://github.com/bengtmartensson/AGirs)?

Oliveirakun, 2017-03-24:
`Sure, Do you really need capturing ("irrecord") running on the RPi? Is in not enough to capture on a PC, say?` 

No, I would like to use a hardware for easy record and send IR commands, and it must to be reliable like (the dead) RedEye and IrTrans. So I need a powerful IR Transceiver and I still didn't find it.
Can this problem of interrupting the CPU every 13 microseconds disturb Ir learning?

I just need to send and record IR commands on command line, I don't need a GUI, that's why I asked about use just IrpMaster. Can I just use it or I need to run Scrutinizer?

`I like a micro processor to do the timing, possibly also decoding. Possibly this, a few components (cf this) used with Agirs?`

It's an arduino shield on top of raspberry pi, so with it there's no 13 microseconds cpu interrupt problem, ok? But the IR Receiver is not good to record commands.
I found this shield: 
https://www.cooking-hacks.com/documentation/tutorials/control-hvac-infrared-devices-from-the-internet-with-ir-remote/
With this shield I don't have the 13 microseconds problem, right?

bengtmartensson, 2017-03-24:
> do you know a good receiver for IR recording?

[Yes](http://www.harctoolbox.org/arduino_nano.html). Note the non-demodulating sensor (TSMP58000). Capturing with a demodulating sensor is like black&white photos: pretty much information is missing...

>  send and record IR commands on command line, I don't need a GUI, that's why I asked about use just IrpMaster. 

Hmm, IrScrutinizer is intended for analyzing etc, not for deployment sending. Check out  ["IrScrutinizer as server"](https://github.com/bengtmartensson/JGirs), and also [IrTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier), which will replace IrpMaster.

> It's an arduino shield on top of raspberry pi, so with it there's no 13 microseconds cpu interrupt problem, ok? 

ok.

Oliveirakun, 2017-03-24:
Ok, I'll take a look on this
Thank you!

bengtmartensson, 2017-03-27:
I am closing this ticket because there is nothing "ticket-wise" to do, and the original question has been answered. Feel free to continue the discussion though.

closed 2017-03-27 by bengtmartensson