bengtmartensson/IrScrutinizer #198: GlobalCache.java - main() usage vs. implementation

neophob, 2017-04-05:
package `org.harctoolbox.harchardware.ir`
file `GlobalCache.java `

in the usage text is written:
```
and command=send_ir,send_serial,listen_serial,set_relay,get_devices,get_version,set_blink,[set|get]_serial,[set|get]_ir,[set|get]_net,[get|set]_state,get_learn,ccf");
```

however the `ccf` is not evaluated as valid command. 
what is the idea? that it would just convert the GC signal to the CCF format? I think this would be handy.

bengtmartensson, 2017-04-05:
There are some main()-functions in particular in package HarcHardware. These were created to test the API therein, and to give some example on their usage. If the main functions are useable in their own right, that is a "bonus feature". Having said that; suggestions and improvements and pull requests  are (at least in most cases) welcome.

The main() function in GlobalCache is basically a simple command line access to most of the GlobalCache TCP function. It can send the commands listed together with appropriate arguments.

If you need to convert a GlobalCache form to Pronto HEX interactively, just past to the "scrutinize signal" window of IrScrutinizer. If you need a program, use org.harctoolbox.harchardware.ir.GlobalCache.parse (that will take a String to an IrSignal), then use org.harctoolbox.IrpMaster.Pronto.toPrintString to convert that IrSignal to the Pronto Hex String.

neophob, 2017-04-05:
thanks @bengtmartensson, make sense. also thanks for your examples, that helped alot. I try to automatically process some payloads (scrutinize them).

However when I try to analyse some payloads, they cannot be decoded - but in the GUI this works fine. Here's an example

```
IrSignal irSignal = GlobalCache.parse("sendir,1:1,1,461000,4,7,96,1356,96,1356,96,7120,96,1356,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,2796,96,4243,96,2796,96,1356,96,2796,96,5686,96,1912");
DecodedSignal[] result = DecodeIR.decode(irSignal.ccfString());
```
the result is an empty array. It works with different payloads with this code, so the setup (jars + jni) should be fine. do you have an idea?

bengtmartensson, 2017-04-06:
The empty array simply means that DecodeIR did not find any decodes, i.e. did not recognize the signal. This is "normal".

`DecodeIR.decode(irSignal)` is better (clearer and more efficient) than `DecodeIR(irSignal.ccfString())`.

neophob, 2017-04-06:
Hmm I also tried the `DecodeIR.decode(irSignal)`with the same result. I ask myself why it works in the IrScrutinizer GUI but not when called with DecodeIR.

bengtmartensson, 2017-04-06:
> it works in the IrScrutinizer GUI 

It does not, at least not for me...

neophob, 2017-04-06:
here's a screenshot http://imgur.com/a/Z2Crc using IrScrunitizer v1.4 - do you use another version?

bengtmartensson, 2017-04-06:
The DecodeIR field is empty. I.e. no decodes.

bengtmartensson, 2017-04-06:
Strictly speaking, DecodeIR and Analyze are both invoked, it is just that the former does not deliver any output.

> what else is done when pressing the scrutinize button?

Parsing, invocation of cleaner and repeatfinder, invocation of DecodeIR and the analyzer, printing in the currently selected text format.

DecodeIr and the analyzer will be replaced by [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) in the future.

neophob, 2017-04-06:
I tried to do a command line version to automatically scrutinize ir signals.

```
	static String convertToGlobalCache(IrSignal irSignal) throws Exception {
		//TODO set repeat count
		return GlobalCache.sendIrString(irSignal, 1, 1, 1, 1, false);
	}

	static IrSignal parse(String payload) throws Exception {
		IrSignal irSignal = GlobalCache.parse(payload);
		return irSignal;
	}

	static IrSignal clean(IrSignal irSignal) throws Exception {
		ModulatedIrSequence irSequence = irSignal.toModulatedIrSequence(1);
        ModulatedIrSequence clean = Cleaner.clean(irSequence, (int)IrpUtils.defaultAbsoluteTolerance, IrpUtils.defaultRelativeTolerance);
        return clean.toIrSignal();
	}

	static IrSignal findRepeat(IrSignal irSignal) throws Exception {
        RepeatFinder repeatFinder = new RepeatFinder(irSignal.toModulatedIrSequence(1), (int)IrpUtils.defaultAbsoluteTolerance, IrpUtils.defaultRelativeTolerance);
        return repeatFinder.toIrSignal(irSignal.toModulatedIrSequence(1));		
	}

	static void cleanup(String payload) throws Exception {
		System.out.println("<< "+payload);
		IrSignal irSignal = parse(payload);

		IrSignal cleaned = clean(irSignal);
		IrSignal result = findRepeat(cleaned);
		System.out.println(">> "+convertToGlobalCache(result));
	}

cleanup("sendir,1:1,1,38000,4,1,97,26,26,26,48,26,48,26,26,26,48,26,26,26,26,26,26,26,48,26,26,26,48,26,48,26,48,26,26,26,26,26,48,26,26,26,26,26,48,26,26,527");

```

the output looks like:
```
<< sendir,1:1,1,38000,4,1,97,26,26,26,48,26,48,26,26,26,48,26,26,26,26,26,26,26,48,26,26,26,48,26,48,26,48,26,26,26,26,26,48,26,26,26,26,26,48,26,26,527
>> sendir,1:1,1,38000,1,43,97,26,26,26,48,26,48,26,26,26,48,26,26,26,26,26,26,26,48,26,26,26,48,26,48,26,48,26,26,26,26,26,48,26,26,26,26,26,48,26,26,527

```
the offset value (43) is definitive wrong, and it looks definitive different than the scrutinizer `sendir,1:1,1,40000,2,1,96,24,24,24,48,24,48,24,24,24,48,24,24,24,24,24,24,24,48,24,24,24,48,24,48,24,48,24,24,24,24,24,48,24,24,24,24,24,48,24,24,528`

Any last hint form you, then I shutup

bengtmartensson, 2017-04-07:
Your results are completely sensible. You are taking one IrSignal (which happens to be a zero intro and a non-trivial repeat), turning it into an IrSequence by taking one copy of the repeat, then invoking the repeatfinder on that. It does not find any repeat there -- how should it? So it turns that IrSequence into an IrSignal with the previous repeat as intro and zero repeat. This is the correct behavior. 

Try changing the "1" in irSignal.toModulatedSequence(1) to something larger, and you will see what I mean...

neophob, 2017-04-10:
Thanks again for your reply. I think we're talking about different issues here - lets start with the easiest:

1. The Global Cache format above is invalid, as the offset 43 points to a non existing part of the payload. I guess thats a problem with the GC exporter.
1. Thanks for your explanation about the logical error I did. I tried to increase the "1" in the . toModulatedSequence call, without any success (well the payload just was bigger). But bascially I want to do the same as the GUI does - bun only as cli call. Which is basically cleaning and find repeat. but I don't really understand what the cleaner effectively does. so what does the cleaner effectively do?

bengtmartensson, 2017-04-12:
1. The spec is somewhat unclear. This is a signal that is not supposed to repeat. I do not consider this a bug or a problem.
2. The cleaner cleans the signal, in that it "removes dirt", random disturbances to the measured durations. A simple-minded way for doing this would be to round the measurements to a multiple of something. I instead compute a histogram, and then bundle similar values to their average. The repeatfinder basically takes IrSequences to IrSignals, by identifying repeated parts in the input  IrSequence. If you are serious about this, I strongly suggest that you take a look at IrpTransmogrifier, where the development take place. It can also be called from the command line.

bengtmartensson, 2017-04-18:
Closing, but feel free to add further comments or questions if you like.

closed 2017-04-18 by bengtmartensson