bengtmartensson/IrScrutinizer #213: IrScrutinizer Mac App

samsonrosen, 2017-06-11:
Hi, 

I have installed IrScrutinizer v1.4 on my Mac - but I cannot see the export option:

Export format: Broadlink-Python IR

Am I missing something?

bengtmartensson, 2017-06-11:
Yes, your are missing something: 1.4 is older than the Broadlink export. You either need to use the [snapshot version](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build), or you can copy [this file](https://raw.githubusercontent.com/bengtmartensson/harctoolboxbundle/master/IrScrutinizer/src/main/config/exportformats.d/Broadlink_Python_IR.xml) to the exportformats.d directory.

samsonrosen, 2017-06-11:
Thanks @bengtmartensson. I am finding that the codes are two short for my Broadlink.

If I take a code such as this from Remote Central:

`0000 0069 0022 0002 0161 00b1 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0042 0016 0016 0016 0042 0016 0042 0016 0042 0016 0042 0016 0042 0016 0042 0016 0016 0016 0042 0016 0016 0016 0042 0016 0042 0016 0042 0016 0042 0016 0042 0016 0042 0016 0016 0016 0042 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0042 0016 0609 0161 0058 0016 00b1`

Then use the "Scrutinise Signal" button. I then export using the Broadlink preset (see screenshot), and am copying from the exported PY file. (see screenshot).

My broadlink software tells me the code is too short. "Input strings must be a multiple of 16 in length".

[image]
[image]

bengtmartensson, 2017-06-12:
Related: https://github.com/mjg59/python-broadlink/issues/90

Original problem/question has been resolved. Fell free to open new issues should the need (or desire) arise.

closed 2017-06-12 by bengtmartensson