bengtmartensson/IrScrutinizer #219: XCF file

Amelichev, 2017-07-28:
Hello! Can you help me with Pronto XCF file? 
http://files.remotecentral.com/view/2447-4550-1/nakamichi_soundspace_10_receiver.html
can't open it with irscrutinizer: "passeexception on line - 1: No Modules element present"
May be you can fix this problem?

bengtmartensson, 2017-07-28:
@Amelichev : Thank you very much for reporting this. The background is that there are (at least) two different version of the xcf format, and the version 5 was never implemented. Stay tuned.

bengtmartensson, 2017-07-29:
Please try the [CI-build](https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build).

Amelichev, 2017-07-31:
Thanks! Yes, this version can open XCF ver 5!

closed 2017-07-31 by Amelichev

bengtmartensson, 2017-08-01:
Thank you for reporting it. Actually, I knew about the problem since the XCF importer was originally written, but never cared to implement version 5 until someone needed it.