bengtmartensson/IrScrutinizer #220: 16-bit NEC Yamaha Protocol Generation

Amelichev, 2017-08-02:
Hi! Can't find 16bit NEC Protocol Generation, or it works not correct. For example this codes:
[image]
Key name "RED" (ID1, ZONE A) = 7A-C0BF
I choose NEC1-f16 Protocol, and try to generati it, byt the generated signal=7A-BEC1
Then I found this page: http://www.awe-europe.com/documents/Control%20Docs/irhex_converter.html
And I had to convert hex to pronto to generate right signal.

May be I'm doing sth wrong?

bengtmartensson, 2017-08-02:
Probably you are interpreting the parameters in the table wrong. It is not very clear... See the screenshot for the parameters to select for "RED". Note that hexadecimal values has to be prefixed by "0x".

[image]

Amelichev, 2017-08-02:
Ок, I genetate 0x7A 0xC0BF:
[image]
But the right signal must be:
[image]

When I capture signal from original remote I see decode IR:
[image]

bengtmartensson, 2017-08-02:
I did not say it explicitly, but exactly what the table means you have to ask its creator. It is by no means clear. 

There are a few [different interpretations in DecodeIR of the parameters in the NEC protocol](http://www.hifi-remote.com/wiki/index.php?title=DecodeIR#Variant_IRstreams_in_NEC_protocols), but this is not synced with the rendering in IrScrutinizer. This will change with the migration to IrpTransmogrifier, cf. #212.

This is not an IrScrutinizer issue.

closed 2017-08-08 by bengtmartensson