bengtmartensson/IrScrutinizer #229: 'Hardware not responding' in IRScrutinizer for Arduino Pro Micro with AGirs installed

JabberwockPL, 2018-03-10:
I am trying to use IRScrutinizer with Arduino Pro Micro with AGirs installed. GirsLite responds correctly when queried with serial console, however in IRScrutinizer I always get the error 'Hardware not responding'. What could be the reason for that?

bengtmartensson, 2018-03-10:
First, turn on Options -> verbose, this will have the communication echoed to the console.

Likely either the serial port or the baud rate is not correct.

JabberwockPL, 2018-03-10:
The parameters are exactly the same as for Arduino serial console (COM3, 115200).

The verbose output is this:
LocalSerialPortBuffered.sendString: Sent '\r'.
LocalSerialPortBuffered.readString: TIMEOUT
LocalSerialPortBuffered.sendString: Sent '\r'.
LocalSerialPortBuffered.readString: TIMEOUT
LocalSerialPortBuffered.sendString: Sent '\r'.
LocalSerialPortBuffered.readString: TIMEOUT

Moreover, even after the error message is closed, IRScrutinizer seems to block the serial port - I need to close IRScrutinizer to be able to open the serial port in the Arduino console again.

bengtmartensson, 2018-03-10:
Exactly what operating system are you using? What driver are you using for the serial port?

JabberwockPL, 2018-03-10:
I am using Windows 7, the module reports as Arduino Leonardo on the driver Arduino LLC ver. 1.2.3.0. GirsLite responds correctly when queried with version, modules, etc. in the Arduino console and in Putty. Of course, I close them before trying to connect with IRScrutinizer.

bengtmartensson, 2018-03-11:
I can now reproduce. It seems like RXTX (the library for serial communication) and the Arduino LLC driver do not get along. 

Related issues: https://github.com/bengtmartensson/HarcHardwareBundle/issues/1 , 
https://github.com/bengtmartensson/HarcHardwareBundle/issues/3

The first one deals with the replacing the outdated RXTX, which should probably be the final solution.

In the meantime, is there another driver that can be used?

JabberwockPL, 2018-03-11:
I understand. As the number of the codes I need is not that great, copying codes and pasting from the console window works for me for now.

bengtmartensson, 2018-03-11:
@JabberwockPL , thank you for bringing this problem to my attention. Since the problem is not solved, I would like to keep the issue open.

The final solution is probably to replace the aged  RXTX, see bengtmartensson/HarcHardwareBundle#1 , Workaround are to use another driver, like a board with a CH34*-chips (see e.g. http://www.harctoolbox.org/arduino_nano.html). Or, to use Linux instead of Windows :-). I am aware that these work-around are not satisfying for all.

JabberwockPL, 2018-03-11:
Sorry, I thought it just duplicated the problem from the other thread. 

Thank you for your assistance!