bengtmartensson/IrScrutinizer #230: IrScrutinizer - Broadlink Python -ir  error

markfal, 2018-04-12:
Hello there, I have  question to ask if I may with regards to exporting via Broadlink Python –ir

When I do and run the exported script  I get the error below .. Cant work out what I’m doing wrong. Any assistance is much appreciated

Regards

Mark

python broadtest.py 
  File "broadtest.py", line 21
    ],
    ^
SyntaxError: invalid syntax

PY script

cat broadtest.py
#! /usr/bin/env python

# This file has been automatically generated by IrScrutinizer.

# For problems, bugs, and suggestions, please open an issue at
# https://github.com/bengtmartensson/harctoolboxbundle/issues

import argparse
import broadlink

IR_TOKEN = 0x26
RF_433_TOKEN = 0xB2
RF_315_TOKEN = 0xD7

commands = {
    # Command #1: On + SAT;     'On + SAT': [
        IR_TOKEN,
        bytearray(),
        bytearray.fromhex('1E5B1E1E1E1E1E3D1E1E1E1E1E1E1E1E1E1E1E1E1E3D1E3D1E3D1E3D1E3D1E3D1E3D1E3D1E3D1E3D1E3D1E1E1E1E1E1E1E1E1E3D1E1E1E3D1E1E1E3D1E3D1E3D1E3D1E1E1E1E1E3D1E0001CD'),
        bytearray()
    ],
}

def get_command_data(command_name, count):
    cmd = commands[command_name]
    buffer = bytearray()
    buffer.append(cmd[0])
    repeat_only = len(cmd[1]) == 0 and len(cmd[3]) == 0
    buffer.append(count - 1 if repeat_only else 0)
    seq = cmd[2] if repeat_only else mk_sequence(cmd, count)
    buffer.append(len(seq) % 256)
    buffer.append(len(seq) / 256)
    return buffer + seq

def mk_sequence(cmd, count):
    no_repeats = count if len(cmd[1]) == 0 else count - 1
    data = cmd[1]
    for i in range(0, no_repeats):
        data = data + cmd[2]
    return data + cmd[3]

def auto_int(x):
    return int(x, 0)

if __name__ == "__main__":
    parser = argparse.ArgumentParser(fromfile_prefix_chars='@')
    parser.add_argument("--device", help="device definition as 'type host mac'")
    parser.add_argument("--type", type=auto_int, default=0x2712, help="type of device")
    parser.add_argument("--host", help="host address")
    parser.add_argument("--mac", help="mac address (hex reverse), as used by python-broadlink library")
    parser.add_argument("--count", default=1, type=int, help="Number of times to send signal")
    parser.add_argument("command", help="Name of command to send, as defined in the code")
    args = parser.parse_args()

    if args.device:
        values = args.device.split()
        type = int(values[0], 0)
        host = values[1]
        mac = bytearray.fromhex(values[2])
    else:
        type = args.type
        host = args.host
        mac = bytearray.fromhex(args.mac)

    dev = broadlink.gendevice(type, (host, 80), mac)
    dev.auth()
    payload = get_command_data(args.command, args.count)
    dev.send_data(payload)

bengtmartensson, 2018-04-13:
Why was this closed? Somehow invalid Python was generated (there is a missing line feed), this is a bug, and should be addressed.

How exactly did you generate that file?; please describe.

markfal, 2018-04-13:
Hi there i closed because  i worked around it
Cheers
Nark

Sent from my SAMSUNG Galaxy S7 on the Telstra Mobile Network

  Python -ir error (#230)

bengtmartensson, 2018-04-13:
I found the problem and have checked in a fix.

Thanx for bringing this problem to my attention, thus helping making IrScrutinizer better.

closed 2018-04-13 by bengtmartensson

markfal, 2018-04-13:
Fantastic  ... 

Sent from my SAMSUNG Galaxy S7 on the Telstra Mobile Network

  Python -ir error (#230)