bengtmartensson/IrScrutinizer #235: Hello, I can not export the remote.

bakdamir, 2018-06-07:
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at org.harctoolbox.irscrutinizer.NamedIrSignal$LearnedIrSignalTableModel.checkName(NamedIrSignal.java:307)
	at org.harctoolbox.irscrutinizer.NamedIrSignal$LearnedIrSignalTableModel.checkCommandSanity(NamedIrSignal.java:315)
	at org.harctoolbox.irscrutinizer.NamedIrSignal$LearnedIrSignalTableModel.sanityCheck(NamedIrSignal.java:299)
	at org.harctoolbox.irscrutinizer.NamedIrSignal$LearnedIrSignalTableModel.getCommandsWithSanityCheck(NamedIrSignal.java:276)
	at org.harctoolbox.irscrutinizer.GuiMain.saveCommands(GuiMain.java:1130)
	at org.harctoolbox.irscrutinizer.GuiMain.saveParametricSignals(GuiMain.java:1111)
	at org.harctoolbox.irscrutinizer.GuiMain.parametricOrRawExportButtonActionPerformed(GuiMain.java:7511)
	at org.harctoolbox.irscrutinizer.GuiMain.access$9900(GuiMain.java:84)
	at org.harctoolbox.irscrutinizer.GuiMain$88.actionPerformed(GuiMain.java:3294)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:290)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2238)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2296)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4897)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4534)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4475)
	at java.awt.Container.dispatchEventImpl(Container.java:2282)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

bengtmartensson, 2018-06-07:
Please provide information on how this can be reproduced.

bakdamir, 2018-06-07:
[image]

bakdamir, 2018-06-07:
[image]
[image]

bengtmartensson, 2018-06-07:
Thank you very much for the bug report. The problem is the empty command names; this is strictly speaking a user error (or sillyness :-)) but the way the program reacts is of course not acceptable.

Should be fixed by commit 9eb58e6, although the the continuous build has (externally induced) problems. Should be resolve at latest tomorrow (I hope...).

Again, thanx for helping me make IrScrutinizer better!

bakdamir, 2018-06-08:
good approach- "(or sillyness :-))"

bengtmartensson, 2018-06-09:
@bakdamir : still waiting for confirmation that your problem was fixed in the recent snapshot.

bakdamir, 2018-06-09:
Yes, corrected!

closed 2018-06-09 by bakdamir