bengtmartensson/IrScrutinizer #237: Why again an error?

bakdamir, 2018-06-08:
[image]

bakdamir, 2018-06-08:
Protocol NEC not known?

bengtmartensson, 2018-06-08:
Yes, the current version does not know "NEC"  [background](http://www.hifi-remote.com/wiki/index.php?title=DecodeIR#NEC) You can manually change to for example "NEC1".

Please check the [current snapshot version}(https://github.com/bengtmartensson/harctoolboxbundle/releases/tag/ci-build).

bakdamir, 2018-06-08:
[image]

bakdamir, 2018-06-08:
does not start

bengtmartensson, 2018-06-08:
Please address ONE thing per issue, and use *sensible* titles.

The missing file should be fixed in a recent commit.

bengtmartensson, 2018-06-09:
@bakdamir : please confirm that the problem with the missing file was fixed in the recent snapshot.

bakdamir, 2018-06-09:
Yes, corrected! I checked it now, thank you.

closed 2018-06-09 by bakdamir