bengtmartensson/IrScrutinizer #239: When there is a correct protocol recognition NEC1&NEC2?

bakdamir, 2018-06-09:


bengtmartensson, 2018-06-09:
The recognitions of the said protocols are correct. Period.

What is your problem?

bakdamir, 2018-06-09:
recognizes as an encoding NECx

bengtmartensson, 2018-06-09:
Some signals do that. Why is this wrong?

Feel free to try [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier).

bakdamir, 2018-06-11:
Is there a graphical interface(GUI)?

bengtmartensson, 2018-06-11:
Presently no, but see #212.

bakdamir, 2018-06-12:
Something does not work
[image]

bakdamir, 2018-06-12:
I do not understand how to run it?

bengtmartensson, 2018-06-12:
```
[1012]$ irptransmogrifier help --short
Usage: IrpTransmogrifier [options] [command] [command_options]
Commands:
   analyze         Analyze signal: tries to find an IRP form with parameters
   bitfield        Evaluate bitfield given as argument.
   code            Generate code for the given target(s)
   convertconfig   Convert an IrpProtocols.ini-file to an IrpProtocols.xml, or vice versa.
   decode          Decode IR signal given as argument
   expression      Evaluate expression given as argument.
   help            Describe the syntax of program and commands.
   lirc            Convert Lirc configuration files to IRP form.
   list            List protocols and their properites
   render          Render signal from parameters
   version         Report version

Use
    "IrpTransmogrifier help" for the full syntax,
    "IrpTransmogrifier help <command>" for a particular command.
    "IrpTransmogrifier help --common" for the common options.
[1013]$ irptransmogrifier help decode 
Decode IR signal given as argument
Usage: decode [options] durations in micro seconds, alternatively pronto hex
  Options:
    -c, --clean
      Invoke cleaner on signal
      Default: false
    --describe
      Print a possibly longer documentation for the present command.
    -R, --dump-repeatfinder
      Print the result of the repeatfinder.
      Default: false
    -f, --frequency
      Set modulation frequency.
    -h, -?, --help
      Print help for this command.
    -i, --input
      File/URL from which to take inputs, one per line.
    -k, --keep-defaulted
      In output, do not remove parameters that are equal to their defaults.
      Default: false
    -n, --namedinput
      File/URL from which to take inputs, one line name, data one line.
    -a, --all, --no-prefer-over
      Output all decodes; ignore prefer-over.
      Default: false
    -p, --protocol
      Comma separated list of protocols to try match (default all).
    --radix
      Radix used for printing of output parameters.
      Default: 10
    -r, --repeat-finder
      Invoke repeat finder on input sequence
      Default: false
    -s, --strict
      Require intro- and repeat sequences to match exactly.
      Default: false
    -T, --trailinggap
      Trailing gap (in micro seconds) added to sequences of odd length.
[1014]$ irptransmogrifier decode 0000 006C 0022 0002 015B 00AD 0016 0041 0016 0041 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0016 0016 0041 0016 0041 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0041 0016 0016 0016 0041 0016 0041 0016 0041 0016 05F7 015B 0057 0016 0E6C
        NEC1: {D=11,F=22}
```

bakdamir, 2018-06-12:
eee @ eee-Lenovo-G570: ~ / IrpTransmogrifier $ irptransmogrifier decode 0000 006D 0022 0000 00AA 00AA 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0478
irptransmogrifier: command not found

bengtmartensson, 2018-06-12:
You should really consult an elementary Linux/Unix text, explaining PATH etc.

Try `./irptransmogrifier ....`

bakdamir, 2018-06-12:
eee@eee-Lenovo-G570:~/IR$ ls
IrpProtocols2html.xsl                              irptransmogrifier.bat  st
IrpProtocols.html                                  irptransmogrifier.sh   xslt
IrpProtocols.xml                                   LICENSE
IrpTransmogrifier-0.2.0-jar-with-dependencies.jar  README.md
eee@eee-Lenovo-G570:~/IR$ ./irptransmogrifier.sh irptransmogrifier decode 0000 006D 0022 0000 00AA 00AA 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0478
Expected a command, got irptransmogrifier

Use "IrpTransmogrifier help" or "IrpTransmogrifier help --short"
for command syntax.
eee@eee-Lenovo-G570:~/IR$

bakdamir, 2018-06-13:
decode 0000 006D 0022 0000 00AA 00AA 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0478
	NECx1: {D=7,F=2,S=7}
	NECx2: {D=7,F=2,S=7}
?

bakdamir, 2018-06-13:
By the way, I patched the folder, but there were 2 identical startup files

bengtmartensson, 2018-06-13:
> decode 0000 006D 0022 0000 00AA 00AA 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 [...]
NECx1: {D=7,F=2,S=7}
NECx2: {D=7,F=2,S=7}
?

however

    irptransmogrifier decode --strict 0000 006D 0022 0000 00AA 00AA 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0040 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0015 0040 0015 0015 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0040 0015 0478

i.e. no decodes with `--strict`. Without `--strict`, empty repeats (like your signal) are considered to match, and thus it is not possible to distinguish between NEC1x and NEC2x. Possibly `--strict` should be the default? But there are so many people who screams "cannot decode my signal" if so.

bengtmartensson, 2018-06-13:
> By the way, I patched the folder, but there were 2 identical startup files

No there is not.

closed 2018-06-14 by bengtmartensson