bengtmartensson/IrScrutinizer #253: Delete selected does not work properly (V. 1.4.2)

irplus-remote, 2018-09-11:
Hi, found a bug..
It deletes the wrong rows unfortunately.

[image]

[image]

bengtmartensson, 2018-09-11:
Thank you very much for this bug report.

I can confirm it: To reproduce: create three lines, select 1 and 3. "Delete selected" will delete 1 and 2. However, I cannot make sense of your stack trace.

irplus-remote, 2018-09-11:
Oh don't worry the stacktrace is from a wrong capture nothing to do with the issue.

And also the "DELETE" key does not work as a shortcut ^^

bengtmartensson, 2018-09-11:
Try the current snapshot; this should fix it.

> And also the "DELETE" key does not work as a shortcut ^^

I claim it does, although it may not be they way you expect it to. The menu has to be visible. This is not to say that the behavior is perfect...

bengtmartensson, 2018-09-13:
@irplus-remote 
Please confirm that the original problem was solved.

irplus-remote, 2018-09-13:
Issue solved ;) (what is strange though is that each code is read with the previous code.  E.g i press on rremote :   1,2,2       , irscrutinizer will capture :  1,1,2  (need to figure out, but it has nothing to do with this one)

closed 2018-09-13 by irplus-remote

bengtmartensson, 2018-09-13:
Glad that it is fixed.

> each code is read with the previous code. E.g i press on rremote : 1,2,2 , irscrutinizer will capture : 1,1,2 (need to figure out, but it has nothing to do with this one)

Sorry, I can't do anything with a description like that. Can be a driver or hardware problem.